prusa3d/Prusa-Firmware-Buddy #654: [QUESTION] Is WIFI supported in latest firmware 4.1.0?

wpegden, 2020-07-07:
The prusa store says that the mini supports wi-fi after upgrading with an esp wifi module.

I have purchased and installed such a module but do not see a way of enabling wifi.  I have just upgraded to the latest firmware and still do not see how to enable the settings.  Nevertheless several of the images in the release notes show a wifi icon in the upper right, so I am hopeful this feature is present.

Is the feature present? How does one activate it?

Thanks.

JohnnyDeer, 2020-07-08:
Hi @wpegden, "future ESP Wifi module upgrade" in our store means that in future we will add this function to firmware. It's not realized yet.  Images are just ilustrative, after implementing it will be described in release notes. Thanks for your question and hope it´s clear now :)

wpegden, 2020-07-08:
Ah okay, thanks! :(

I was looking at the feature comparison table, which I guess is the one place where "future" is not used in the description of the WiFi upgrade.

I guess there is no timeframe currently for the upgrade?

JohnnyDeer, 2020-07-13:
I can't promise any date, but it's on plan. Can we close this issue as solved? :)

closed 2020-07-13 by wpegden