prusa3d/Prusa-Firmware-Buddy #655: [BUG] Update on usb drive is not recognized

jokohoko, 2020-07-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-   4.1.0-RC1

**Original or Custom firmware** Original

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
I have release candidate firmware 4.1.0 RC1 on my prusa mini and i tried updating to the recently released 4.1.0 final version. But the printer does not seem to recognize the update on the usb drive.
The usb drive is recognized as it should but the printer does not show a message to start the update.
All previous updates in the past worked as expected.
I tried multiple usb drives, so think the issue is in the printer or update rather than the drive.

**How to reproduce**
Try to update from 4.1.0-RC1 to 4.1.0

JohnnyDeer, 2020-07-08:
Hi @jokohoko please check settings in your printer about updates. Settings - FW Update - Always ON, it may be reason of your issue.

ricotrot, 2020-07-08:
Hi. 
I seem to have a similar issue here.
The only difference is I am on 4.0.5 and it is my first update.
Tried multiple USB drives, even reinstalling 4.0.5 - the printer doesn't seem to recognize the update on the disk.
I tried to go Settings - FW Update - ...but when I click on FW Update, I can very briefly see Loading, then the printer reverts to home screen.

jokohoko, 2020-07-09:
It did work after changing the setting and a couple of resets.
Is there a reason this setting could have been at off by default? I am testing all version as they come out and would have never set this settings to off myself... (or it would have been by accident).

nicolasvac, 2020-07-10:
I can confirm the problem happened to my printer too.

To be able to update, i had to reset the printer with the USB already inserted.
But when i click on FW update in the settings, it flashes for half a second the logo and then goes back to the home screen.

JohnnyDeer, 2020-07-13:
@ricotrot @nicolasvac this is just misunderstanding how "FW Update" works. Choice "Always" mean that printer will update after POWER ON if FW is on usb stock. "On restart" mean that printer will update after restarting printer if FW in os usb stick. Printer will not start updating after click in menu.

@jokohoko setting shoud not be changed by default. So updating works after change "FW update" in menu right?

nicolasvac, 2020-07-13:
@JohnnyDeer there is no choice on that option. As soon as you click it, it shows the loading screen and takes you back to the home screen.

JohnnyDeer, 2020-07-13:
@nicolasvac yes on version 4.0.5 and less. Since 4.1.0-RC1 is this chois in menu. Either way FW Update funcion will not update printer after click in menu but after reset or after power on. If is FW Update set to OFF updating printer stays as before.

nicolasvac, 2020-07-13:
@JohnnyDeer ah ok, gotcha. 
Thank you for claryfying.

jokohoko, 2020-07-13:
@JohnnyDeer Yes, after changing these settings the update dit run correctly. Only on all my printers the default setting is off for both settings? And thus both did not update without changing the setting. (one early release prusa mini and one of a few weeks ago). Maybe other people should verify...

JohnnyDeer, 2020-07-14:
@jokohoko good, so there is no bug. We will look into default settings. I´m closing this issue as invalid, because it´s not a bug. Thank you for reporting.

closed 2020-07-14 by JohnnyDeer