prusa3d/Prusa-Firmware-Buddy #660: [FEATURE REQUEST] Enable LCD display timeout

nicolasvac, 2020-07-08:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]

**Is your feature request related to an existing problem? Please describe.**
I keep my printer in my room, and having the LCD display always up not only is irritating during night, but uses power for nothing, and consumes the display life.
Plus, in a farm of printers, the power problem gets multiplied by much.

**Describe the expected functionality**
  Add in the settings an option for display timeout for inactivity, like 1 minutes, 5 minutes and so on.
The display simply comes back on by pressing on the button 1 time.

yesBilly, 2020-07-15:
I came here to request this also. Thanks @nicolasvac

neophyl, 2020-07-16:
I would modify the request slightly in that any input from button or rotating the encoder should wake the device up.

nicolasvac, 2020-07-16:
@JohnnyDeer any feedback on this?

JohnnyDeer, 2020-07-16:
@nicolasvac I don't discuss with developers about this FR yet. But definitely it make sense :)

adynis, 2020-07-24:
I also vote for this.
I would even like to have 2 different timeout options: Timeout while printing and Timeout when idle (e.g.: the printer finished to print and it's waiting for ... me).

JohnnyDeer, 2020-07-27:
I have updated info. We will make this kind of function but i can't say when. Anyway it's on our to do list :)

hemdrup, 2020-12-19:
Yes, this would be great. And then just having to move the button to wake it up. 

However, as there is no LED indicator for when the Prusa is on I might use the power switch instead as I don't know that the display is asleep/power saving.

adynis, 2020-12-19:
you are right with the LED indicator ...
than maybe something like a screensaver with a dot moving on the screen .... or the date/time moving slowly on the screen with low brightness ....[similar to AlwaysOnDisplay on samsung mobile phones]

PGlyre, 2021-02-13:
I also want this function and would appreciate that the LED screen is turned off or/and dimmed e.g. 50% after 10 minutes when using OctoPrint etc.

calexandre, 2021-02-14:
I also would like to be able to set a display timeout. It does not make sense for it to stay on when there is no activity...

hainjedaf, 2021-05-24:
Well, i'd like choices:
-screen on with timeout
-screen off
-screensaver
maybe something where the menu shows an option to turn off screen and wake it by turning the jog dail.

matthewlloyd, 2021-07-13:
I attempted to control the backlight from the firmware and did not have success. The backlight LEDs are not be under the control of the ST7789V. Sending WRCTRLD and toggling the backlight bit (ST7789V_MASK_CTRLD_BL in the code) does not have any effect, and sending WRDISBV doesn't adjust the backlight brightness but instead the brightness scaling value of the LCD pixels.

Examining the [PCB file](https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-display-ldopr28lcd-a003.zip) with [this viewer](https://www.altium.com/viewer/), and cross referencing the [TFT LCD datasheet](https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-lcd-module-fw024tft-v14.pdf), reveals that the backlight LED is indeed permanently powered from the 5V supply and not under control of the ST7789V or mainboard.

Therefore, I think it's safe to say turning off the backlight to save energy isn't ever going to be possible on the Mini, unless hardware changes are made to the LCD board.

Displaying all black or running a screensaver would definitely be possible, but likely not as useful.

adynis, 2021-07-13:
> Examining the [PCB file](https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-display-ldopr28lcd-a003.zip) with [this viewer](https://www.altium.com/viewer/), [...]

Nice investigation :thumbsup: but sad conclusion :worried:

Fliens, 2022-08-04:
> I would modify the request slightly in that any input from button or rotating the encoder should wake the device up.
With the new prusalink, the screen should be waken up if a print is started trough prusalink

hainjedaf, 2022-08-07:
Hi Simon,

I think that is the idea but i just had not thought of how to implement.
Your suggestions sound natural way to do

Kind regards
Marout Sluijter-Borms

Prusa-Support, 2022-08-17:
Hello! Thank you for your feedback and sorry for the late reply from my side. Unfortunately, even after consulting with devs about this, we found the LCD likely doesn't support this and, like @matthewlloyd said, drawing all the pixels black won't save energy. Leaving the thread open for now in case there might be new findings about this, but likely it sadly won't be implemented given this info. 

Alessandro Pantaleo
Prusa Research

rovili, 2023-08-04:
Any update on this?

I think nowadays, with sustainability all over the place; this is a must-have feature. Also, adjusting the bridgeness of the screen would be a great feature. (I am light sensitive and the white font on the black background is very bothering for me)

GRIFFCOMMca, 2023-08-21:
Came here to suggest LCD brightness the same as the Mk3 has, timeout would also be nice or be able to set the brightness to 10% so its super dim and not burn the screen.

mpapierski, 2023-11-28:
Any update on this? This would be a great feature for always-on printers.

DeFo162, 2024-03-06:
My Mk4 stand in the basement and I don't want to draw the neighbors' attention to the printer.

t-makaro, 2024-03-06:
I sleep near my mk4. The sound doesn't bother me, but the light does. I had to print a cover for the LCD which is annoying to have to remember to put on all the time. Bonus points if I could disable some of the other LEDs too (LED status bar already disabled).

csweichel, 2024-03-12:
> Hello! Thank you for your feedback and sorry for the late reply from my side. Unfortunately, even after consulting with devs about this, we found the LCD likely doesn't support this and, like [...]

@Prusa-Support is the same true for the mk4 as well, considering it's an all new display? Seeing how long this issue's been open for, how high is the chance we'll see this coming in a firmware update? If so, what would be a timeline to expect?

adynis, 2024-03-12:
From my point of view, even if energy is not saved (because of the background light which is still on) I prefer to have this option because the LCD light is ... annoying during the night .

GRIFFCOMMca, 2024-03-13:
Kinda surprised you dont have brightness control on the backlight LED... however it appears many want the display to be off as its in a living area, so making the pixles BLACK may not save power, but the goal is to dim / make the display go off, so making the LCD black would be a valid fix.

Prusa-Support, 2024-05-24:
Thanks for your feedback.

The old MINI and the new MK4/XL don't natively support this feature but our developers were making considerations about this request recently. It turned out that a workaround would be hardly feasible at the current time and unfortunately, I have no details about the technical limitations. 
Sadly, I don't have a final answer but we will keep collecting feedback and possibly review with the developers in the near future.

Covering the LCD as some users do is surely an opinion if you find the screen to be annoying at night.
However, I strongly encourage you not to print in the vicinity of people - or pets! - at night, and to physically turn the printer off via the rocker switch every time the printer is not in use. Sleep-wellness shouldn't be underestimated and please note the LCD power consumption is negligible; in fact, the printer like most electrical devices, draws power even when idle.

Michele Moramarco
Prusa Research

murk-sy, 2024-07-30:
Although brightness cannot be controlled, just blacking out or significantly dimming the screen would likely likely reduce lighting up rooms at night unnecessarily - even if there's nobody there.

Having it activate during stealth mode (with timeout) would make a lot of sense, especially so if it's scheduled stealth mode.

t-makaro, 2024-08-27:
I took the stl files for the Mk4 display housing and used that to make a super basic cover for the screen. It's a super basic model that I hadn't bother to upload until [now](https://www.printables.com/model/987702-mk4-display-cover). Hopefully it helps some of you.

bkerler, 2024-09-01:
PR #4174 does allow to control the display backlight control. I will also look into the LCD display timeout.

lanwin, 2026-09-21:
^^ but its still relevant

samdurnxn, 2026-11-13:
In the file src/leds/led_manager.cpp there is the comment : LCD backlight is connected to the green channel of the fourth LED (index 3) on the status strip. The function  LEDManager::set_lcd_brightness()  is only use at the initialization to set it to 100.  Maybe it will be great to add a new command in connect API to manage it.