prusa3d/Prusa-Firmware-Buddy #671: [ENHANCEMENT] Tell user to install build sheet before self test

jokohoko, 2020-07-12:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI
**Optional upgrades** - none

**Is your suggested improvement related to an existing problem? Please describe.**
I had quite a bit of issues getting the printer to pass the self-test. No clue what solved the issue in the end, but halfway all of my tests i was running the self-test without a build sheet installed. The interface never mentions if this is required or not, but it will not pass without the sheet installed. Only after quite a while i installed the sheet and it passed immediately. 

**Describe the expected functionality**
Like the MK printers, ask the user whether a sheet is installed or not. And tell the user that a sheet is required before the temperature tests.

x10102, 2020-07-12:
Great idea! Took me a few minutes to figure it out for the first time.

Eldenroot, 2022-11-26:
It happened to my once when I bought the printer... some confirmation or notice on the display would be nice.

Prusa-Support, 2022-12-23:
The heaters test changed shortly after this issue was posted, with FW [4.3.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.0-RC1), as the selt-test heaters thresholds were modified.

Basically, for the heating test to pass the self-test:
- the nozzle range should be 130-190°C after 42 seconds;
- the bed range should be 50-65°C after 60 seconds.

This said, specific ambient conditions might affect the result of the self-test more than the sheet itself.

I'm wondering if you experienced this inconvenience again with FW 4.3.0 (or newer), in "regular" ambient temperatures (18-38°C as per the manual).
If not, it may be safe to say that this issue is solved.

Michele Moramarco
Prusa Research

closed 2024-07-04 by github-actions[bot]