prusa3d/Prusa-Firmware-Buddy #677: [FEATURE REQUEST] Power panic

hawwwran, 2020-07-14:
**Printer type** - [MINI]

I thought power panic is MINIs feature already. Found the hard way it probably isn't.

I expect the same behavior as MK3s has. Rise the print head and continue once the power is restored.

npgoalkeeper, 2020-07-15:
I'm not a dev, however I doubt that in the case of a blackout there would be enough residual power remaining in the (small) PSU that the mini uses.

derenma, 2020-07-16:
@npgoalkeeper It's listed as a future software upgrade here: https://www.prusa3d.com/original-prusa-mini/

yonie1, 2020-08-27:
I looked at the schematic an the pin 'BED_MON" could be used to detect a falling supply voltage and cut heater power immediately if detected an save the position. But I think this will sadly never be as good as the mk3 that can detect a power outage on the primary side and therefore has a larger Power reserve to work with because the primary side caps are still charged

d0ugak, 2020-11-16:
Considering the power supply that is supplied with the mini doesn't have the same hardware as the power supplies in the i3. The caps that can keep enough power stored to safe the printer and store its last status before the power is cut. There is probably no way to implement this exactly as it is on the i3's. If they did implement such a feature it would probably be something along the lines of constantly writing either to eeprom on the buddy board or a "status file" on the thumbdrive which file it is reading from and which line in the file it last processed. (lots of wear and tear for flash memory) That way if power was cut, it could read from this eeprom or "status file" to resume where it left of. My guess is it would not result in the most seamless of resumes depending on where in the tool movement power was lost after the last line read of the gcode file. There is also the issue of the still hot print head just sitting on the part as it cools causing imperfections in the print. Perhaps it might be possible to detect power loss on the power supply as the voltage rail starts to sag and immediately cut power to the heaters to use the little residual power left in the power supply caps to jerk the print head up off the piece being printed. But then there is the issue of truly knowing where the print head is once power is restored. You could program it in that jerk movement to jerk the print head up 1cm, but if there was only enough residual power that the print head only moved up 0.5cm and then on power resume the printer just assumes to move the print head down 1cm and resume printing, it will slam into the piece being printed.

Probably the best way to implement power panic on the hardware that the mini has, is build in support to the firmware to read the status of a USB connected UPS that powers the printer. On powerloss the firmware would be able to detect the UPS is in offline mode, safe the printer, write status out to eeprom or a "status file" on the thumbdrive then pause the print, waiting for either the UPS to go back online and resume. or to be freshly powered back up after power is restored from an extended outage and then resume.

SgtKiLLx, 2021-09-06:
This needs to be added sooner rather then later. I know I'm not special, or the only one in dire need of this. We all have a legit reason for why we need this.. but come on. Any day now

HabibiHany, 2022-04-15:
Just wanted to keep bumping this up. I know this was promised and i know its on the bucket list but here is hoping this comment will help!

ShadowDrakken, 2022-05-20:
Any movement on this? I thought the Mini's control board was supposed to be bigger, badder, more advanced than the older models? Maybe with an addon board?

Prusa-Support, 2022-05-27:
Hello! Sorry for replying so late to this thread. Thank you for your feedback on this aspect, unfortunately, we don't have info on this at this time, but I brought this to our devs attention for them to review. 

Alessandro Pantaleo
Prusa Research

wesleyjg, 2022-06-13:
Well, I didn't realize the Mini DIDN'T have power panic until just now. A storm knocked out power mid print and now I have to start over.

Gordontgopher, 2022-06-14:
Coming 'soon'? https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2170

Prusa-Support, 2022-06-22:
Hello! Sadly, nothing confirmed on a possible timeframe for now @Gordontgopher, but hopefully it will be all OK there and it can be released! 

Alessandro Pantaleo
Prusa Research

jhenley01, 2022-07-21:
My experience with the Mini has definitely lowered my opinion of Prusa, it's a pretty good printer as is but we're coming up on three years since launch and still don't have the features that were promised. No wifi, no power loss recovery, no print farm software, no firmware updates since December. I'm guessing they put the Mini development on the back burner so they could work on the XL.

Prusa-Support, 2022-07-22:
Hello! Thank you for your feedback and I'm sorry these aspects made your experience with the MINI negative. I hope the future developments, though no timeframe at this time yet, can make up for the experience and improve it, I absolutely understand these features were expected for a long time. In case of anything to further review in more detail, feel free to let me know at any time, here on the comments or directly with me info@prusa3d.com, mentioning me. 

Alessandro Pantaleo
Prusa Research

DWardFRC1329, 2022-09-19:
This feature is definitely still sought after! The Prusa Mini has been a fantastic printer for us, but this one feature is a blemish on an otherwise good printer. You still have users yearning for this feature. Don't forget about us!

Niflheim123, 2022-09-20:
I am seeing this :
https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2170 
POWER_PANIC
Yuri D'Elia [wavexx@thregr.org](mailto:wavexx@thregr.org)

Is it related?

Prusa-Support, 2022-10-04:
Hello, I confirm, it is related, to indicate the feature is assigned to our dev, though unfortunately we are still unable to confirm a possible timeframe for this feature at this very moment.

Alessandro Pantaleo
Prusa Research

xXxLorenzo85xXx, 2023-02-09:
Is it related POWER_PANIC? Thanks

Prusa-Support, 2023-02-18:
Hello @xXxLorenzo85xXx, I confirm it, it is related with power panic, though we unfortunately don't have yet a timeframe for this, sorry for not having more info on this aspect. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-08-17:
Sadly no news at the moment.
Unfortunately, the implementation of this feature proved to be extremely tricky.
The various SW solutions that we tested so far proved to be unreliable or even hazardous.

We will not release a feature that is not considered to be totally safe but this won't prevent us from keeping on trying. 

Michele Moramarco
Prusa Research

ShadowDrakken, 2023-08-17:
> 

Maybe don't advertise features before they're implemented in the future then eh? Don't promise things your hardware is incapable of. This feels like a given, and it feels like something that should have been done during product development, not after release. We are not your beta testers.

oneteam2win, 2023-12-19:
It's unbelievable how Prusa has been behaving without any sense of accountability. The feature in question was a crucial factor in my purchasing decision years ago. Particularly for prints lasting hours, a power interruption of just a few seconds can be incredibly frustrating. I feel let down by Prusa as they promised a feature that, even after years, remains unavailable. To me, this feels like a breach of trust.

Noxmiles, 2024-01-28:
@Prusa-Support Can you give an adress for Legal inquiries? You lied, many customers (including me) are not getting what you promised over years. You really have to compensate buyers for this fraud.

np22-jpg, 2024-01-28:
> @Prusa-Support Can you give an adress for Legal inquiries? You lied, many customers (including me) are not getting what you promised over years. You really have to compensate buyers for this fraud.

@Noxmiles I don’t think anyone here has the serious resources required for litigation, unfortunately, and suing over a minor feature that failed to got shipped is kinda iffy. Who knows, PR might be able to get it out eventually. Firmware has always been a sore spot for them.

Original webpage (since the old one was changed and had references to “software-based power panic” removed):
https://web.archive.org/web/20191016153722/https://www.prusa3d.com/original-prusa-mini/

oneteam2win, 2024-02-09:
@Moramarco
I'm sorry but I don't understand the irony of the feedback.
As far as I'm concerned, I was one of the first buyers of the Prusa Mini and got a printer for a lot of money that was almost unusable. Further investment in the direction to an Mini+ made the printer halfway usable. Unfortunately, however, an important selling point for me (power panic) is still not available.
But as a thank you, we get an ironic response.
Very nice, thank you very much.

moramarco, 2024-02-09:
> @moramarco I'm sorry but I don't understand the irony of the feedback. As far as I'm concerned, I was one of the first buyers of the Prusa Mini and got a printer for a lot of money that was almost [...]

Ciao a tutti. Rispondi in italiano perché in realtà il vero proprietario dell'account di GitHub è italiano, ed è la persona che sta scrivendo ora. 

A quanto pare qua qualcuno si diverte a utilizzare il nome di miei familiari, utilizzando il mio profilo GitHub aperto anni fa e mai utilizzato realmente se non per prendere spunto per qualche progetto scolastico. Non ho idea di cosa stiate parlando...ho letto qualcosa inerente stampanti ma non ho intenzione di perdere ulteriore tempo a leggere tutto il thread, non interessandomi. 

Sappiate che qualunque commento è stato fatto sulla chat con il mio account è stato fatto da qualcuno che ha bucato il mio account GitHub. Ho provveduto a eseguire solo ora che mi è arrivata la mail che mi avvisava del vostro commento e che mi ha fatto capire questa storia. 

Procederò penalmente per capire gli autori di sta cazzata.

Scusatemi, buona giornata a tutti

closed 2024-07-04 by github-actions[bot]