prusa3d/Prusa-Firmware-Buddy #699: [BUG] Z Axis crash if Load filament when head is near the top of z axis

simonbugler, 2020-07-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
Mini
**Printer firmware version**-  [e.g. 4.0.5, ...]
2.1.0

**Original or Custom firmware** [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
None

**USB drive or USB/Octoprint**
USB

**Describe the bug**
When the head is at the top of the z axis and you load filament. The head moves up 3-4 cms and crashes into the top.

**How to reproduce**
  Please describe steps to reproduce the behavior.
Place head at or near the top of the z axis and load filament.

**Expected behavior**
Before moving up - it should do a homing procedure so it knows its at the bottom, or knows its height.

**G-code**
None

**Crash dump file**
None

**Video**
None

JohnnyDeer, 2020-07-21:
Hi @simonbugler I also confirm this bug. Thank you for reporting.

jokohoko, 2020-07-23:
Could an advantage of the 32bit board be to save a rough height on shutdown?
So that when you change the filament it has some sense of the height it's at.

I think saving the height is somewhat how they plan on doing power loss recovery. so it might help here as well.

JohnnyDeer, 2020-07-27:
We noted this bug and at this moment have low priority. Thank you for understanding.

jokohoko, 2020-07-27:
Do these Trinamic drivers also have the current control?
Maybe you could lower the current during these moves so that the stuttering is less loud.

zoltan-l, 2020-09-01:
this is always some "trade off" technically there would not be a problem at each switchon to run the auto home procedure, but it takes time and  energy. Someone could also argue - when you save the coordinates during switch off, someone can move a Z manually after it, so when you switch on the machine, you can never be sure, the coordinates (even from eeprom) are correct. SImplest workaround is - when you see the Z is to high run auto home.

Fliens, 2020-12-14:
I just had the same bug in 4.3.0-RC1
It scared the hell out of me when my z axis just made some loud noise and didn't had some kind of detection that it reached the top!

abrugh, 2021-02-28:
The only time this really happens to me is when I'm changing nozzles. I move the print head to the top, swap the nozzle then load new filament. I did this, just now, right after a print - the printer knew where the head was, why is it blindly ramming upwards for 5 seconds? It would be fantastic if this could be addressed.

JohnnyDeer, 2021-09-06:
We considered this is not bug which need to be fixed for right function of printer. So this issue have low priority. It can be avoided by lowering Z height or autohome before loading/unloading filament. Even if it happen printer won´t be damaged. Thank you for understanding.

Prusa-Support, 2022-05-18:
Hello! Sorry for intervening so late on this thread. I have to inform that this aspect won't be fixed, I'll close this issue based on this info. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-18 by Prusa-Support