prusa3d/Prusa-Firmware-Buddy #705: [ENHANCEMENT] Prusa mini codes on RepRap wiki

thinkyhead, 2020-07-22:
I see in the Prusa mini code that there are a few [new or re-assigned](https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/src/marlin_stubs/pause) G-codes, such as `M603` to configure filament change. On the RepRap wiki this G-code is only documented as a "[Stop print](https://reprap.org/wiki/G-code#M603:_Stop_print)" (similar to `M0` in RepRapFirmware). However there is no mention or documentation for the new overloaded meaning of this code.

It's important to document these codes at some point —preferably before they are deployed— so that other firmware which uses the RepRap wiki as a reference can decide which codes to use for new functions. It may also be helpful to contact some of the active firmware projects in order to confer about G-code assignments so that there is something like a standard.

I had a chance last year at 3DMS to have a long discussion on the issue of G-code standards with David Crocker and Tony Lock of Duet 3D and Gina Haüßge of OctoPrint. Everyone at the table was keen to try and get a standards document together. The basis for this would be the current documentation at the RepRap wiki, but would also look to the future and consider things like the new G-code variables and compare-and-loop stuff which has been incorporated into RepRapFirmware.

JohnnyDeer, 2020-07-23:
Hello @thinkyhead thank you for suggestion. We agree with your point and  answer is yes we have plan to update specific g-codes used on mini. However there won´t be much of them used on mini :)

3d-gussner, 2020-07-23:
Updated https://reprap.org/wiki/G-code#M603:_Stop_print 
This isn't final and needs to be updated later.

3d-gussner, 2020-07-23:
Maybe https://reprap.org/wiki/G-code#M703:_Configure_Filament would match better than a new M603
In RepRap M703 "hich may contain codes to set parameters like extrusion factor, **retraction distances** and temperatures"

closed 2024-03-03 by thinkyhead