prusa3d/Prusa-Firmware-Buddy #706: [ENHANCEMENT] Isn't heat bed heating unnecessary when loading/unloading filament?

fuelflo, 2020-07-22:
**Printer type** - [MINI]
**Optional upgrades** - [Filament Runout Sensor]

**Is your suggested improvement related to an existing problem? Please describe.**
no

**Describe the expected functionality**
When loading or unloading filament while NOT printing the printer heats up nozzle and the heat bed. Often heating up the heat bed is even slower and you'll have to wait for it to reach 60°C.

Now I'm no expert, but isn't heating the heat bed on a lone filament change/load/unload unnecessary? First of all it would save a lot of energy if only the nozzle would be heated and also it would save a bit of time because the printer wouldn't need to wait for the bed heating.

jokohoko, 2020-07-23:
I don't think the printer waits for the bed temp to start the filament change process.
And in a lot of cases the next step after changing materials will be actually starting a print.

So i would agree with the current logic that also heating the bed is the most useful method. A menu option to "also heat up bed" or "cooldown bed" in the change material process could be useful in some situations.

fuelflo, 2020-07-24:
Hey,

I checked again and the MINI might not be waiting till the bed heating is finished, when only loading/unloading filament.

But when only **unloading** (like when you don't plan to use the printer for a specific time) it still heats the bed, which IMHO is a waste of energy and life time of the heater. When unloading, to load another filament after that, the heating still can be done in the filament loading process.

However, even when **loading** filament to print right after, the printer heats the nozzle to >=215°C (depending in the material) and the bed to >=60°C after loading and purging the printer needs to wait and cool down the nozzle temp to 170°C (to prevent oozing of filament during mesh bed leveling I guess). So we could use the cool down time of the nozzle to heat the heat bed for people who want to print right after. And for people who don't want to print the bed would never have been heated unnecessarily.

jbnjohnathan, 2021-03-08:
I agree that when you select "unload filament" the reason is often because you are not using the printer for a while. And as said before, if you do want to print no time is lost if the bed would be heated when loading the filament.

oerkel47, 2021-11-02:
I agree. This Behaviour is super stupid.

Additionally: When choosing "Unload filament" it should set temps to 0 after. Otherwise the differentiation with "change filament" doesn't make sense.

Gyrxiur, 2021-11-02:
Thanks, it really needs to change!

RichiH, 2021-11-04:
In a Prusa Live episode many many moons ago, @josefprusa asked that we send customer support requests in case a GitHub issue is really biting us. I just contacted support asking for an ETA and will report back with what I get.

RichiH, 2021-11-05:
Reply from support:

> I've spoken to the devs in regard to the issue before, and the current statement is that the heating is intentional, as when you are loading/unloading filaments, it's extremely likely that you would using the printer very soon, and having the heatbed pre-headed should be a convenience more often than not. I have submitted your feedback though for further consideration, as I' see you've commented on GitHub too already, and this goes a long way. Ideally for me, it would be great to have this as a setting that can be toggled.

jbnjohnathan, 2021-11-08:
> I've spoken to the devs in regard to the issue before, and the current statement is that the heating is intentional, as when you are loading/unloading filaments, it's extremely likely that you would [...]

As mentioned here before, there are three options. "Load Filament", "Unload Filament" and "Change Filament"
In my opinion when "Change Filament" is used, it's highley likely that you will print shortly thereafter, and thus warming the bed could be useful. Same if you use "Load Filament".
But when you use "Unload Filament" its rather highely likely that you will not print afterwards since you are removing the filament and not replacing it with anything. When using this option heating the bed is not neccessary.

tcurdt, 2022-03-15:
> as when you are loading/unloading filaments, it's extremely likely that you would using the printer very soon

This is what engineers usually call a premature optimisation. I'd also question the data on the likeliness.
I could live with a load/unload/change or even better a toggle in the settings.
But the current behaviour is nothing less but annoying.

Please keep it KISS. A heated bed is just not needed for loading/unloading filaments.

ThomasCZ, 2022-09-09:
Please fix this issue, it is wasting of electricity.

Prusa-Support, 2022-09-13:
Hello! Thank you for the request and for the comments, sorry for the late response from my side to the post. All brought up to our devs attention, options are being reviewed by them. In case of anything from our side, we'll update here, in case of anything more you'd like to add for us to review, do not hesitate to post more comments. 

Alessandro Pantaleo
Prusa Research

RichiH, 2022-09-13:
@Prusa-Support thanks for the update.

As a summary "please let us choose a default we like" is what we're asking for.

Optionally being able to abort the heating in case it's set to on but we don't want it this time would be nice as well. In the latter case, telling the user about the new setting in the UI might be nice.

acollaborator, 2023-04-30:
I would like this behavior change too.  

The use cases are unloading filament to change a nozzle or a filament as well as loading filament. 

I do not need and do not want the bed heated for these use cases.   I would like one less potential "that's hot" jump-scare in my day.

I do not starting printing within seconds of these use cases.

ScottWell1, 2023-07-04:
Most concern seems to be with the bed heating, but I also find the nozzle staying hot after "Unload Filament" to be problematic.  

If I was changing filament to print again soon, I would use "Change Filament".  But when I use "Unload", I'm typically done printing for the day and want the nozzle to cool down after Unload is complete.

As an alternative, it would be nice to at least have an additional "Cooldown" option on the Filament menu, so it can be manually selected after unloading, without first returning Home and going to another menu.

Patronics, 2023-08-26:
Yet another vote for either disabling this or allowing it to be configurable, it's so wasteful, and now I'm stuck waiting for the bed to cool again so I can move the printer, which was the reason for unloading the filament to begin with.

OneOfTwelve, 2023-10-05:
It would be a great thing if the bed didn't heat up when changing filament.
For one of the things I use my MINI for most, I don't need the bed heated, but I have to change the Filament quite often.
I can turn heating off while printing but as soon as I change the filament the bed heats up and I have to wait until the bed is cold again. A toggle "Preheat bed when changing filament" would make my life a lot easier.

fuelflo, 2023-11-02:
> It would be a great thing if the bed didn't heat up when changing filament. For one of the things I use my MINI for most, I don't need the bed heated, but I have to change the Filament quite often. [...]

This option now seems to be available on 5.1.0 alpha firmware for buddy.

RichiH, 2023-11-02:
For the XL, I can confirm that there is a toggle between just nozzle and
nozzle&bed in the alpha.

Sent by mobile; please excuse my brevity.

Prusa-Support, 2024-06-14:
I'm sorry, I believe this has already been fully implemented.

You can follow the path
`Settings -> User Interface -> For filament change, preheat [Nozzle/Noz&Bed]`
and set your preference.
https://help.prusa3d.com/article/lcd-menu-mk4_413171

Michele Moramarco
Prusa Research

closed 2024-06-21 by Prusa-Support