prusa3d/Prusa-Firmware-Buddy #708: [BUG] Estimated end time not matching between MINI Printer and Prusa Connect

DangeMask, 2020-07-23:
**Printer type** - [MINI]

**Printer firmware version**-  4.1.0-final+1923 original FW

**Printing from USB drive**

**Describe the bug**
  When starting a 13+hour print today morning, the printer was showing estimated time around 21:00 in the afternoon. When I checked after a few hours, the estimated time was 5 o'clock tomorrow morning. But the estimation in Prusa Connect stays the same.

Based on percentage and first estimation, I think the printer estimate is wrong somehow, but not by exact hours (my first theory was wrong timezone calculation).

[image]
[image]

The time zone setting in printer are UTC+2 (I am located in Czech republic).

I have upgraded to 4.1.0 yesterday and printed a few "longer" prints, but didn't notice this behaviour.

I can upload the gcode tomorrow, as I don't have a copy on my computer. Need to wait for print to finish.

JohnnyDeer, 2020-07-23:
Hi @DangeMask thank you for reporting. Gcode will be helpful.

DangeMask, 2020-07-23:
Just an idea: attaching 3MF project from PrusaSlicer 2.2.0. The settings are the same as currently printing gcode, so the output should be the same.

But I will also attach the original gcode when the print finishes.

[SpiralYarnBowl.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4965527/SpiralYarnBowl.zip)

DangeMask, 2020-07-23:
UPDATE: continuously checking the print and I also had to pause the print for a while, so I have a new info for debugging.
The estimated time on printer is changing from time to time... so it is not stuck on some error, the printer still updates the time, just not correctly :-)
It would be handy for this to see current time in printer menu, just to check if it is fault of NTP or fault of estimation.

DangeMask, 2020-07-23:
UPDATE: Now even the day of end is "Today". If I ignore the "date" info, the difference between printer and web estimation would stay the same. But I don't understand the change from "Friday" to "Today" after 30min pause.
[image]

DangeMask, 2020-07-24:
UPDATE: Original gcode
[SpiralYarnBowl_0.2mm_PLA_MINI_13h27m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4970478/SpiralYarnBowl_0.2mm_PLA_MINI_13h27m.gcode.zip)
I'll keep you updated if the issue continue

DangeMask, 2020-07-24:
New print, starting 08:50, estimated time of print 1hour 32 minues
Printer says ETE 11:32, web 10:22 (web looks ok to me)
UTC set +2
I must admit, I forgor to restart the printer before this print. That's unforgivable, I am IT guy and I forgot to try "turn it off and on again" :-(
[image]
[image]
[image]

DangeMask, 2020-07-24:
1 hour later and the ETE is different only by 2 minutes. *mind blown*
[image]
[image]

Vic-Y, 2020-07-25:
I am seeing some similar symptoms with my MINI running firmware 4.1.0. Printer UI is one hour different from WebUI. Time zone is UTC-7 (Mountain Time) and daylight savings.

[image]

[image]

[image]

AcidWeb, 2020-08-07:
Definitely still occurs in 4.2.0. Plus I think I had some NTP related BSOD.
[image]

Vic-Y, 2020-08-08:
Firmware updated to 4.2.0.

Screen estimate said print ending at 04:38 (4:38 am) - but print actually ended at 21:38 (9:38pm).
[image]

DangeMask, 2020-08-10:
Testing today with FW 4.2.0. When print started, both printer and Web showed ETE 08:15 CEST. Now I just chgecked the printer directly and it shows "Today at 00:16", which looks like remaining time (15 minutes until end) instead of time of end.
[image]

DangeMask, 2020-08-14:
Looks like at the start of print, the difference in EOP between printer screen and web is a few minutes and around 50% of print, the printer EOP time changes dramaticaly to something totaly out.
Were you able to replicate this issue in testing or is it just a "one in a milion printers" error?

stixpjr, 2020-08-15:
Also saw this on 4.2.0-final+2067. At the start of the print, the end time times matched and were realistic, when I came back 30 minutes later, the times were 10 hours out. First test print with 4.2.0, and 0.05mm layer height.

[image]
[image]

anderscallertun, 2020-08-16:
I'm almost always getting the wrong EOP on the screen and it's off by many hours and sometimes has the wrong day.
It's correct at first, but after an hour (possibly an hour after the last reboot) the EOP gets screwed up. The EOP on the Prusa Connect is always correct. Is the printer possibly updating the time using NTP after an hour and gets it very wrong then?

Example:
Starting a 4h23m print at 2:04. EOP at start is "Today at 6:31" (correct). After 1 hour EOP jumps to "Today at 14:31" (very wrong).

Without restarting the printer I at 18:17 that day started the same print. EOP was correct with 22:41, but after 39 minutes the printer displayed a "WATCHDOG RESET" message and stopped printing.

At 21:41 I pressed continue on the error and started the same print with an EOP of "Today at 10:04" (very wrong).

I then restarted the printer at 21:44 and started the same print again. EOP was "Sun 02:09" (correct), but after 1 hour it jumped to "Today at 10:09" (very wrong).

It seems the error is actually always 8 hours, but depending on when on the day it can be the wrong day too.

I have set TZ to UTC+2.

Timelaps of most prints in my example:
[timelaps.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5079762/timelaps.zip)

Vic-Y, 2020-08-16:
I'm seeing similar symptoms as @anderscallertun.

At start of print Sat 21:00.
Print duration according to PrusaSlicer is 4h 58m and therefore print should finish Sun 01:58. 
Printer displays EOP as Sun 00:56 and PrusaConnect displays EOP as Tomorrow (Sun) 01:59.

At 20% Sat 22:00
Printer displays EOP as Today 08:58 and PrusaConnect displays EOP Tomorrow (Sun) 01:59 - unchanged.

blysmth, 2020-08-16:
I didn't notice any issues with the EOP times until I updated to v4.2

Most prints are displaying EOP times that have no relation to the expected time, sometimes many hours ahead. If I go into the tuning menu and change the TZ, it generally brings it _closer_ to the expected time but just now I'm toggling between UTZ and UTZ+1 and selecting +1 is consistently adding 2 hours to my EOP time.

JohnnyDeer, 2020-08-17:
Thank you all for informations and Gcodes. We are working to resolve this issue.

DangeMask, 2020-10-10:
Interresting new detail: When I change print speed, the EOP time on screen changes to sooner value, but web EOP stays roughly the same and slowly decreases.
During writing this comment, it went from 20:25 to 20:18 on website and from 19:49 to 19:51 on screen.
[image]
[image]