prusa3d/Prusa-Firmware-Buddy #717: [BUG] Entering the tuning menu while octoprinting screws up head movement.

d4rken, 2020-07-26:
**Printer type** - [MINI]

**Printer firmware version**-  v4.2.0-RC1

**Original or Custom firmware** [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint** Octoprint

**Describe the bug**
While printing via OctoPrint, if I enter the tune menu, the head moves up and out of the while (while extruding), and then moves back to the print, printing in the air.

**How to reproduce**
Start print via OctoPrint, enter tuning menu on the printer. In some cases also reproducible while bedleveling.

**Expected behavior**
Headmovement not being interrupted by entering the tuning menu.

**G-code**
It's not specific to a print.
[GEPRC_Mark4_BN-880Q_GPS_Mount_v2_0.2mm_FLEX_MINI_1h16m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4977527/GEPRC_Mark4_BN-880Q_GPS_Mount_v2_0.2mm_FLEX_MINI_1h16m.zip)

**Crash dump file**
Didn't crash.

**Video**
If you can't reproduce it, I'll make one, currently don't want to break my ongoing print.

JohnnyDeer, 2020-07-27:
Hi @d4rken did you noticed this issue on 4.1.0 or there was no problem?

d4rken, 2020-07-27:
I have not tried `4.1.0`.

d4rken, 2020-07-27:
Doesn't happen on the `4.1.0` release.

JohnnyDeer, 2020-07-27:
@d4rken thank you for find out.

RobeeeJay, 2020-08-05:
I have this happening to me as well on `4.2.0-RC1`, it hasn't happened on `4.1.0` or any earlier release that I know of. It specifically happens when I adjust Live Z up or down. It instantly ruins a print, but thankfully I am only doing this on the first layer.

RobeeeJay, 2020-08-12:
@d4rken have you noticed this happening in the release version of 4.2.0? I've changed settings a few times now in the Tune menu and it doesn't seem to be happening to me any more. Maybe it was fixed in final?

d4rken, 2020-08-12:
I have not had the time to test this. I'd have expected this issue to be mentioned in a commit or PR if it was fixed?

Chleba, 2020-08-12:
Hi,
This has been fixed in FINAL release version 4.2.0.

JohnnyDeer, 2020-08-17:
I can confirm it's fixed. I'm closing this issue as solved

closed 2020-08-17 by JohnnyDeer