prusa3d/Prusa-Firmware-Buddy #730: [FEATURE REQUEST] Filament status override

x10102, 2020-07-29:
Sometimes, for example, when the filament breaks in the extruder and I have to remove it manually, the printer thinks filament is loaded when it really isn't. It's annoying because I have to press "Unload filament", wait for the printer to unload nothing, and then I can actually load it.

I was thinking: When the user wants to load filament when one is already supposed to be loaded, the printer could display a prompt saying something like "Filament is already loaded, do you really want to proceed?" so that you wouldn't need to uselessly wait until the printer "unloads".

JohnnyDeer, 2020-08-04:
Hi @x10102 thank you for suggestion. We have plan to implement this kind of feature.

x10102, 2020-08-06:
@JohnnyDeer Thanks! Should I close the issue or not?

JohnnyDeer, 2020-08-13:
@x10102 rather leave it open till this feature will be implemented :)

hemdrup, 2020-12-19:
Yes. Some kind of cancel or restart override feature would be great with loading filement. 
Sometimes it has not grabbed the filament when inserting it and then I have to wait wait until I finally get to say NO it's not the right color and then can try and insert the filament again.

JohnnyDeer, 2021-05-20:
@x10102 feature already implemented, can you please confirm it and close thi issue?

closed 2021-05-21 by x10102