prusa3d/Prusa-Firmware-Buddy #733: [BUG] After print finished it does a homing & purge sequence but no line

Bjerg, 2020-07-29:
**Printer type** - 
Mini

**Printer firmware version**-  
4.1.0

**Original or Custom firmware** 
Original

**Optional upgrades** - 
Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  After print finishes it homes and makes the purges line move but does not purge only happens after upgrading from 4.0.5 dont think i saw the behaviour in 4.1.0 RC1 does not matter what slicer i use

**How to reproduce**
  I upgraded to 4.1.0 and it made my Mini crash into stuff after it finished printing. so i do not know why or how to reproduce the error

**Expected behavior**
  Its suppose to just do end gcode but it doesnt

**G-code**
 Do not think its gcode related since i have changed nothing in the endscript

**Crash dump file**
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4995716/dump.bin.zip)

**Video**
  https://youtu.be/8hS-jPmPzk4

malcmcmreal, 2020-07-29:
Hi, I have a very similar issue except the printer starts the mesh bed leveling sequence instead of purge. I'm also on 4.1.0, I tried 4.2.0 and still happens there. Never happened 4.0.5

See my post on the help forums including a video:
https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/immediately-after-successful-print-mini-goes-into-mesh-bed-leveling-procedure/

zoltan-l, 2020-08-05:
Bjerg, Malcolmcmreal
can you please attach your gcode? The best wpuld be the .3mf project file.

zoltan-l, 2020-08-05:
seems malcmcmreal´s trouble was the G80 at the end part of the G-code  which is another way of asking for MBL

Bjerg, 2020-08-05:
Yes of course here is my gcode. its only happening with 4.1.0 and up

[Shape-Box_0.2mm_PLA_MINI_16m (3).zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5027846/Shape-Box_0.2mm_PLA_MINI_16m.3.zip)

malcmcmreal, 2020-08-05:
> seems malcmcmreal´s trouble was the G80 at the end part of the G-code which is another way of asking for MBL

I've replied in the Prusa forums (https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/immediately-after-successful-print-mini-goes-into-mesh-bed-leveling-procedure/#post-240376).

I tried without the G80 and my issues persisted. By the way, the reason I added the G80 in the first place was to try and stop the steppers from moving post print. The problem was there before my G80 addition. 

My printer has changed it behavior again, it not longer trys to mesh bed level but now just immediately returns to home post-print. It moves the Z and Y axes at the same time. This happens with and without the G80.

Similar to Bjerg, I also have another extruder (Bondtech) with same 415 estep.

Here are my files:
[test print.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5028121/test.print.zip)

zoltan-l, 2020-08-06:
I am investigating, as all of you has inserted user G-codes, and from G-code is not that simple what everything you have changed - stay tuned

zoltan-l, 2020-08-06:
Gentlemen, can you do me a favour? Post here all G-code changes and places where you inserted you custom G-code?

malcmcmreal, 2020-08-06:
Hope this helps, I've only changed my start and end gcode in slicer. Changes I've made are in **bold** 

START GCODE
G90 ;   use absolute coordinates
M83 ; extruder relative mode
**G28 ; home all
G1 X100 Y100 F4000
G1 Z1.5 F50 ; park position
M301 P17.10 I1.25 D58.60 ; PID   values tuned with RHD_B fan and Mosquito @ 220C**
M104 S170 ; set extruder temp   for bed leveling
M140   S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed   leveling temp
M190   S[first_layer_bed_temperature] ; wait for bed temp
**G4 S120; wait 2 mins**
G28 ; home all without mesh   bed level
G29 ; mesh bed leveling
M104   S[first_layer_temperature] ; set extruder temp
G92 E0.0
G1 **Y0.0** X179 F2400
G1 Z3 F720
M109   S[first_layer_temperature] ; wait for extruder temp
**M92 E415 ; for Bondtech   extruder**
 
 
; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110.0 E8.0 F900
G1 X40.0 E10.0 F700
**G1 E-0.8 F2100; anti-stringing   retract
G1 X41.0 E0 F1000.0 ; -1mm   wipe
G1 X36.0 E0 F1000.0 ; 5mm line   @ 0
G1 E0.6 F1800; detract 80%**
G92 E0.0
 
M221 S95 ; set flow
 
 
END GCODE
G1 E-1 F2100 ; retract
{if layer_z <   max_print_height}G1 Z{z_offset+min(**layer_z+15**, max_print_height)}{endif} F720 ; Move print head up
G1 X178 Y180 F4200 ; park   print head
M84; stop the stepper motors
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow
**M220 S100 ; reset speed factor**
M900 K0 ; reset linear  acceleration

zoltan-l, 2020-08-06:
Bjerg,
does this happen also with g-code without M92?

Bjerg, 2020-08-06:
@zoltan-l just tried it only happens with M92 enabled without it does not. on 4.1.0 and up

zoltan-l, 2020-08-06:
@Bjerg 
thanks,  this will help much, we suspected it - forwarde to development team - stay tuned

FrenzyIncarnate, 2020-10-12:
This issue also happening with my printer. Only G-code change that I ve made is adding "M92 E415 ; set e-step for bondtech extruder".
Mine reacts a bit different tho. Goes to X178 Y180 as usual then goes to X 0 Y180 Z0 and stops there after the print ends. I will also be looking forward of dev teams solution

FrenzyIncarnate, 2020-10-17:
Hello. I've found a way to prevent this until there is a fix. As we know Bondtech gives us 2 solutions 
1 adding that line to Start Gcode at Slicers printer settings
2 they provide a Gcode file that does not print just sets required Estep value.

When I start my printer first I run that Gcode . By this way even if I swap filament or do a first layer calibration it does not underextrude and finish the prints as it should be. I have removed the M92 E415 from Start Gcode

I don't know if people figured this already but just wanted to share it.

ModelTrainNut, 2020-12-08:
Multiple M92 axes on one line will trigger the problem on roughly half the prints on my MINI: 
M92 X100.4 Y100.6 Z100.8

I separated the axes onto multiple lines and have not seen the problem for weeks:
M92 X100.4
M92 Y100.6
M92 Z400.8

hamlinjacob, 2020-12-19:
Has this been resolved with the new release candidate? I am having the same issue with 6 of my minis. All have the bondtech upgrade.

hamlinjacob, 2020-12-19:
The bug has not been resolved in the new firmware update. I just flashed one of my minis and ran a test print. The same issue occurs.

malcmcmreal, 2020-12-19:
Maybe we should add that installing the Bondtech requires the M92 but also requires swapping some wires on the buddy board. See steps 33-34  https://support.bondtech.se/Guide/01.+Feeder+Upgrade+Kit/59?lang=en

maybe that has something to do with it

hamlinjacob, 2020-12-19:
I am not sure why swapping extruder wires would cause the end Gcode to
lower to the back left hand corner.

The other print you sent me finished. It did not move to the back left hand
corner. So it’s likely a bug related to M92 and the firmware. Apparently an
earlier firmware did not have the issue.

ModelTrainNut, 2020-12-19:
I have the factory extruder on my MINI.

When I added the M92 command, with multiple axes on one line, the problem
started for me:
M92 X100.6 Y100.8 Z401.0

I then split it into multiple M92 commands, one for each axis, and the
problem has never happened since:
M92 X100.6
M92 Y100.8
M92 Z401.0

Randy

derenma, 2020-12-19:
@hamlinjacob Are you using a USB extension, by chance? I had to remove that as well as replacing my USB stick and my issues went away. Both the stick and extension work fine for normal PC use, but I think there is something else involved that I am missing.

ModelTrainNut, 2020-12-19:
I am using the original silver plated usb that came with it with no
extension cord.

hamlinjacob, 2020-12-19:
I am not using a usb extension. I am using a different thumb drive though.
This is also happening through multiple minis. After removing M92 E412 the
issue isn’t there.

Does anyone know what the normal esteps for the mini is? Can we overcome
the Estep issue by increasing flow or extrusion multiplier?

ModelTrainNut, 2020-12-19:
I've never used Set Axis Steps for the Extruder on either my MINI or MK3:
M92 Ennn

I use Set Flow Percentage instead: M221 S103.26 in my case, copying how the
factory MK3 gcode tweaks it to 95% flow to hide an old glitch in Slic3r.

Randy

zoltan-l, 2020-12-20:
Guys,
the issue has been discovered. It is not a M92 issue.  The and the team is working on it. It is caused by the comment line which is longer than 95 characters and is created by Slicer from Custom Gcode added to slicer and contains Gcodes. Workaround: enter your gcode file and delete this long comment line containg Gcodes.
The bug will be resolved in the next release.

derenma, 2020-12-20:
@zoltan-l So I am only partially crazy? NICE! 

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/864

derenma, 2020-12-20:
I am not going down that rabbit hole again. 🤣

hamlinjacob, 2020-12-20:
I just added a pause to the end of the print file before end gcode and it is another easy workaround.

hamlinjacob, 2021-02-14:
The issue still isn't fixed with the latest firmware update and slicer update :(

derenma, 2021-02-14:
I think this is the PR for the fix: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1217

It's going to be in a future release soon, hopefully.

zoltan-l, 2021-02-16:
Correct

JohnnyDeer, 2021-06-07:
@Bjerg this issue should be fixed, can you confirm it and consider this issue to close?

FrenzyIncarnate, 2021-06-07:
Hello guys. I believe Jo Prusa said that e-step values will be adjusted via screen. When this will be implemented?

Bjerg, 2021-06-07:
@JohnnyDeer its working now no more weird behaviour after print finishes ill close the issue for now

closed 2021-06-07 by Bjerg