prusa3d/Prusa-Firmware-Buddy #738: [BUG] Nozzle can crash into the bed aligment bolts

omegahelix77, 2020-07-30:
**Printer type** - [MINI]

**Printer firmware version**-  4.0.5

**Original or Custom firmware** Original

**Optional upgrades** - None

**USB drive**

**Describe the bug**
  When printing a large object, near the limits of the machine, the nozzle hits against the print bed alignment bolts when printing the skirt around the perimeter of the object. I would expect the firmware or perhaps Prusa slice to prevent this.

**How to reproduce**
  Print my object on a Prusa Mini.

**Expected behavior**
  Nozzle does not hit anything.

**G-code**
[https://drive.google.com/file/d/1tyK61v9RdjRdRfMBHdGnRn9Ufq7QE6Ne/view?usp=sharing](url)

**Crash dump file**
  N/A

**Video**
  Maybe later

derenma, 2020-07-30:
That is some extreme printing. I just physically tested that on my Mini and the only way I could get the nozzle to hit the beds was manually after disabling the motors.

When Y is at the full 180 position (which is just a few mm short of the physical maximum of Y), the nozzle does not make contact. It's super close, but it doesn't bump anything on my machine.

omegahelix77, 2020-07-30:
Thank you for testing out my issue. I did not realize I was doing some extreme printing. I knew I was near the limits of what the Mini could do but I really would not expect it to be possible for the nozzle to hit the bed alignment bolts. I wonder why it happens on mine and not yours. My mini came assembled and I have not made any modifications. My guess is that Prusa Slicer checks that the object is within boundaries but does not regard the skirt. That, or it doesn't take the bolts into account. Maybe I could cut them down a bit, they don't need to be as long  as they are.

derenma, 2020-07-30:
I don't want to speak for any Prusa rep here, but this sounds like a physical misalignment of some kind.

You could try turning off the skirt for large prints as a quick fix, I suppose.

JohnnyDeer, 2020-08-06:
Hi @omegahelix77 thank you for reporting. We are researching this issue.

JohnnyDeer, 2020-08-19:
@omegahelix77 can you please check which version is your belt holder? You should be able to see it just trough place between two bearings on left side of bed. We have made version F1 and F2. Thank you.

omegahelix77, 2020-08-19:
There is an "F1" on the bottom. See pic.
[image]

JohnnyDeer, 2020-08-26:
@omegahelix77  I meant this part https://github.com/prusa3d/Original-Prusa-MINI/blob/master/STL/MINI-y-belt-holder.stl .  If you have F1 version on your MINI please print and replace it for posted R6 version and let me know i it helps, thank you.

derenma, 2020-08-27:
On this note, what are the risks of using R1-6 versions instead of 'F' versions? @JohnnyDeer

Slower iterations on Factory parts is totally cool, and understandable. I am hesitant to replace factory parts, since, well, they are from the factory.

JohnnyDeer, 2020-08-27:
@derenma We are constantley improving our printers therefore some parts can have more versions. Parts marked F are printed on our farm, parts with R are available to everyone to print. There is no difference in design between F and R and there is no risk in using newer version printed at home.

s-taylor, 2021-01-20:
I've had this exact same issue trying to print at 18cm in length. I ended up trying to adjust any ZX axis skew by loosening and tightening the bolts, as well as adjusting my x-axis belt tension and skew. I tried to get the Z axis perpendicular to the bed using a right angle. This did help a lot, and my prints are now much more centered, but it still does just clip the screws at the back.

You can read all about my troubles here https://forum.prusaprinters.org/forum/assembly-and-first-prints-troubleshooting/z-banding-issues-2

JohnnyDeer, 2021-02-17:
@omegahelix77 is this issue still relevant? 
@s-taylor did you tried to change Belt Holder to actual version as mentioned up here?

JohnnyDeer, 2021-05-06:
I´m closing this issue as solved and stale.

closed 2021-05-06 by JohnnyDeer