prusa3d/Prusa-Firmware-Buddy #739: [BUG] Filament sensor doesn´t work with octoprint "Serial port"

pocharlies, 2020-08-01:
**Printer type** - [MINI]

**Printer firmware version**-  4.1.0 Final

**Original or Custom firmware** Original

**Optional upgrades** - Filament Sensor Runout

**USB drive or USB/Octoprint** With USB Drive works with USB/Octoprint doesn´t work

**Describe the bug**
Filament sensor is not working when octoprint is printing.

**How to reproduce**
Just try to print in serial port and filament sensor changes are not detected

**Expected behavior**
Filament sensor detects and M600 Command is launched in Octoprint

nwalke, 2020-08-03:
This is being fixed in #503 I believe.  I think it's being added after the current work around localisation.

JohnnyDeer, 2020-08-05:
Hi @pocharlies thank you for reporting. We already know about this bug.

andreas4all, 2020-09-02:
still in 4.2.1

RichiH, 2020-12-11:
@JohnnyDeer Is there any timeline to a fix for this issue? Just lost a print of almost a day and spent roughly an hour trying to fix the resulting jam without disassembling the extruder at first, and then disassembling the extruder after all.

DaReaLDeviL, 2020-12-18:
Same problem here like @RichiH - already messured the meters with the spool before print and maybe must about only 2 meters off :( . Please do a fix for that.

buildcomics, 2021-01-09:
Stumbled into the same on this end. I considered hooking up the filament sensor to octoprint using this plugin: [https://plugins.octoprint.org/plugins/filamentrevolutions/](https://plugins.octoprint.org/plugins/filamentrevolutions/)
However, I am running octoprint on a server pc that does not have GPIO. I have an arduino attached, but that won't work without modifying the plugin to take a script as detection input. So for now, I am waiting for this to be fixed, so the printer can trigger it's own filament change procedure even when connected through a serial port.

Yith1, 2021-01-09:
> Stumbled into the same on this end. I considered hooking up the filament sensor to octoprint using this plugin: https://plugins.octoprint.org/plugins/filamentrevolutions/
I have it working plugged into my octopi with this plugin. https://plugins.octoprint.org/plugins/filamentsensorsimplified/ 
It works really well.

buildcomics, 2021-01-09:
> > Stumbled into the same on this end. I considered hooking up the filament sensor to octoprint using this plugin...
> 
> I have it working plugged into my octopi with this plugin. https://plugins.octoprint.org/plugins/filamentsensorsimplified/
> It works really well.

Thank for the tip. Didn't know there was (yet) another plugin there. This plugin still assumes a raspberry pi though, which I don't have. I have a x86 server PC running, and along with octoprint running multiple applications. I like the power of it (much more powerfull than a raspi), but lacks GPIO. A script option would enable me to attach an arduino (which is already there for PSU control) such that I can use that same arduino for my filament sensor connection. But actually fixing this issue would solve the entire problem and I wouldn't have to re-wire the Prusa sensor from the printer to something else and leave the printer "untouched".

cdrage, 2021-01-09:
> > Stumbled into the same on this end. I considered hooking up the filament sensor to octoprint using this plugin...
> 
> I have it working plugged into my octopi with this plugin. https://plugins.octoprint.org/plugins/filamentsensorsimplified/
> It works really well.

Do you know what pins specifically should be used with the Prusa Mini filament sensor? I'm having troubles figuring out what goes where.

Yith1, 2021-01-09:
> > > Stumbled into the same on this end. I considered hooking up the filament sensor to octoprint using this plugin...
> > 
> > 
> > I have it working plugged into my octopi with this plugin. https://plugins.octoprint.org/plugins/filamentsensorsimplified/
> > It works really well.
> 
> Do you know what pins specifically should be used with the Prusa Mini filament sensor? I'm having troubles figuring out what goes where (there are three cables).

I posted a guide on facebook. I hope that you can see it. 
https://www.facebook.com/groups/prusaminiusers/permalink/3784090601654259

buildcomics, 2021-01-09:
> I posted a guide on facebook. I hope that you can see it.
> https://www.facebook.com/groups/prusaminiusers/permalink/3784090601654259

It is a closed group, but I got in. Hope you don't mind me posting the connection(s) here for reference:

[image]

Crowlord, 2021-01-18:
any chance of a fix on this. If I had known that the filament sensors were totally useless on an Octoprint setup I would not have bothered buying them. Its now over a year later and a version update but STILL this basic functionality is missing. 

I dont want to have to rig up a fix to gpio i want the firmware to be finished.

ZeekWorld, 2021-01-18:
+1 I'm on the same boat as Crowlord described above.

qwertycody, 2021-01-22:
+1 have same issue - looking to not change standard wiring or hookup on ports unless absolutely necessary

bergmal, 2021-01-24:
+1 i also have this probem... i thought prusa was capable of fixing such a basic error

danielo515, 2021-01-26:
So, if prusa mini really wants to be a printing powerhouse this is a very needed feature. Instrumenting the printer from a third party controller is important for any serious printing workload (and even just amateur!!!). What's the point of having a sensor that you can not actually use?

kingjamez, 2021-01-30:
How is it possible that this still isn't fixed. Lost my print last night because I believed the filament sensor would... well.... work.

mix579, 2021-02-05:
I just discovered this issue the hard way... print failed when filament ran out and the sensor didn't trigger. It clearly sees if filament is in or out when loading/unloading but didn't trigger when job was run from my raspi. Totally defeats the purpose of a filament sensor...

mik-at, 2021-02-07:
Bug still there in 4.3.0 pls fix soon :(

Dimonstar, 2021-02-17:
Desperately need this functionality as well, please fix.

chiefspecialk, 2021-02-28:
Really need this fixed as it's causing major issues losing prints without it.

magarto, 2021-03-02:
Is there any news about this bug?

Crowlord, 2021-03-05:
Prusa Mini starts shipping Nov 2019....

It is now 15 months later, mini has been updated to mini+

AND THE DAMNED FILAMENT SENSOR STILL DOESNT WORK WITH OCTOPRINT.

This is Anycubic levels of inept.

sayeghr, 2021-03-07:
This is important, please fix!

refob, 2021-03-09:
This bug still exists in FW 4.3.1+2759 - please consider to fix it ASAP!

My Prusa Mini+ is connected to a Raspberry Pi running OctoPrint 1.5.2.. I tested the filament sensor using the debug menu and it seems to work fine. I also ran a second test and could verify that the printer does not stop when the filement exits the sensor. This not acceptable, because in the end 360 mm of filement are left inside the tube blocking the insertion of new filament.

janis-ax, 2021-03-24:
+1 Same issue

tomishnet, 2021-03-27:
... me too! I was expecting some alert, but nothing. Please fix this issue or extend internal server of Prusa mini for files uploading and printing. In this case is filament sensor only stupid(and not cheap) gadget.

samwscott, 2021-04-19:
+1 please fix ASAP!  

The filament sensor I purchased with my Mini is completely useless without this functionality.  Ugh, I just wasted 3/4 spool with a long print.  How can all the other sensors output data via serial but not the filament sensor?!

Crowlord, 2021-04-19:
Pretty sure at this point there is no way to fix this bug. That or they just dont care.

ZeekWorld, 2021-04-19:
I would say they don't care.

samwscott, 2021-04-19:
Well, that's disheartening.  I've only been doing this (3d printing) for a week.  I chose Prusa because I thought they cared and worked to fix software issues.  After a few test prints I got the courage to do a 30 hour print only to lose it at 95% because a filament sensor designed and sold to prevent this problem failed to do its job.  And not because of the hardware; because it's a software bug!  Why is this so hard to fix?

danielo515, 2021-04-20:
We don't know if it is hard or easy to fix because anyone at  prusa didn't even took the time to answer at all. They are too busy building new stuff and leaving behind the previous ones, just like any Chinese manufacturer

mik-at, 2021-04-20:
It looks like someone tried to fix it in https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/503 but gave up early :(

Joshi425, 2021-04-20:
just talked to Pedro from Prusa Customer Support, he said Support for this will come in the next Firmware Release

blejdfist, 2021-04-20:
I for one is longing for this to be implemented in the firmware. But please people, there can be many reasons why the developers have not implemented this yet. I'm sure they will get to it when they can.

As a developer myself it's disheartening to see people calling other developers names or questioning their intelligence when I'm sure they are doing their best to provide a solid product we can all be happy to own. 

So please try to keep a respectful tone.

Crowlord, 2021-04-20:
I did. For over a year. Its time they fixed it. I have lost too many prints
to give benefit of the doubt now.

rasssta, 2021-04-20:
I would really love this feature. But if you print a long print right now, just use a filament manager (there are plenty in OctoPrint's plugin directory) so you know if it'll work out or not. It's not that hard.

ZeekWorld, 2021-04-20:
octoprint users are a minority, Prusa may be focused on profit, its going to be two years since the release of the mini, and Im still waiting for the wifi feature Prusa promised with the printer release... I would suggest you all to find a workaround like I did for the filament ranout, unless a third party open dev comes up with some programming I doubt we will get the fix anytime soon.

Crowlord, 2021-04-21:
@rasssta it is when you have a printer farm and close to 30 rolls in different colours.  Thats why we have the mini and octoprint after all and thats what it is sold as.

tibbe, 2021-04-21:
Can you please stop spamming everyone subscribed to this bug, who are waiting for updates on the actual issue, please.

Crowlord, 2021-04-21:
Ironic that you have become the very thing you swore to destroy.

On a side note, now you have spammed my inbox by complaining, perhaps if
theres enough life in the thread Prusa will actually listen and fix what is
essentially a completely unforgivable bug in a printer that is specifically
sold as a print farm.

danielo515, 2021-04-21:
> But please people, there can be many reasons why the developers have not implemented this yet. I'm sure they will get to it when they can.
> As a developer myself it's disheartening to see people calling other developers names or questioning their intelligence

People may be mentioning developers, but I never saw anyone writing disrespectful comments or any people name. This was just generic `developers`mention.
Apart from that, people are mentioning developers because they think they are the ones fixing this, and ultimately they are, but prioritising this and giving the attention it deserves is company's responsibility. So what people are actually complaining about it's company priorities, even if they mention a generic team of developers. No developer should ever be facing this thread, some responsible should and, when they decide it's time, they will give a task to a developer who will fix it.
And as a company I think they are doing this very wrong, and us, as paid customers with unmet expectations are in our right of complain. Someone at Prusa could have answered, peace people, ask them for patience, or maybe give a rough implementation date, no matter how far. Instead, they decide to keep not answering nor fixing this bug, and people get mad about it, and I think they should.

I am also a developer, and while I take responsibility of my work, I don't feel any bad when a customer complains about a bug that my company think it's not worth prioritising. I do my work, they do theirs and we are all adult and professional.

ZeekWorld, 2021-04-21:
@Joshi425 

"Vote for Pedro"

mix579, 2021-05-08:
I just heard them say in the latest Prusa Live video that a fix will come in the next firmware release.

ZeekWorld, 2021-05-08:
can you share the video link please?

matthewlloyd, 2021-05-08:
"... and the filament sensor fix, working with OctoPrint, should be in 4.32."

RichiH, 2021-05-08:
Wondering if /me mentioning this issue in the live stream before last had
anything to do with speeding things up :p

Sent by mobile; please excuse my brevity.

fhb, 2021-05-08:
Did they give an idea when the update can be expected?

RichiH, 2021-05-08:
Not that I remember, only "soon".

Sent by mobile; please excuse my brevity.

chrishilbert, 2021-05-09:
Lost 2 prints today due to this bug.  Luckily it was the filament snapping off at like 5 layers in so not much materials lost.  Hoping for a fix soon.

JohnnyDeer, 2021-05-13:
This issue will be fixed in upcoming FW version 4.3.2. Thank you all for patience.

JohnnyDeer, 2021-05-25:
New FW version with fix of this bug is out. Please try it out ad let know if this issue can be closed :) 
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2-RC1

Crowlord, 2021-05-25:
This appears to be working. 

Thank you.

dmyers7, 2021-05-25:
Agree, filament run out while using OctoPrint now triggers a filament change on the Mini directly and the print resumes once addressed.  Not overlay sure, but it may have dropped the gcode commands in the que before the pause.  That said the resumed print is perfectly registered and I can not detect and visual issues; my simulated run out was on and interior fill.  For those that want to try it, download the Release Candidate from the github release page.  It is not on the prusa website yet.  Remember RC releases may still contain bugs and are not officially released till everything is tested.

JohnnyDeer, 2021-06-01:
I'm closing this issue as solved, thank you all for help.

closed 2021-06-01 by JohnnyDeer

kevinwlau, 2021-07-23:
Wasn't sure whether to open a new issue, but wanted to report that the filament change _did not_ trigger for me when the end of my filament passed through the sensor.

Running 4.3.2-RC1+2775. Not sure if RC2 made any changes to this, but the changelog didn't mention it.

mix579, 2021-07-23:
Same here. Doesn't trigger when running remote jobs via Astroprint. The Astroprint team checked the log files on the raspi, no sign of anything. The filament just runs out, just like before.

bschollnick, 2021-08-13:
Small Issue here.

IT appears that Octoprint will not detect the missing filament, if the print is started without filament in the sensor.
eg.  You are remote printing, and forgot to load the filament before you started the print.

My assumption is that it's expecting a change in status to indicate the loss of filament?

Otherwise, when I tested with Filament loaded, and then pulled it out, it **did** detect the loss of the filament.