prusa3d/Prusa-Firmware-Buddy #754: [QUESTION] png2cc.sh and Languages

Kar-Mar, 2020-08-05:
Hi guys, I am a medium programmer and I need little help. I have 2 problems with compiling my mini firmware.

First of all, I want to change the graphic interface (icons, etc.) - how to convert the PNG to compatible C arrays (i tried several algorithms but it will not show me my icons. Any guidance? png2cc script cannot be run - has trouble with Upper cases and other issues

Secondly, I want to modify my firmware to show different languages - no problem to translate a few words. But again how to convert PO translations to Binary form of raw and string files? https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/lang/README.md  I tried this Readme but at the end of the file, it says "anyway we did it differently" and does not tell how.

I tried to talk with my fellow programmers but no one understands this unclear documentation. 

Best regards

DRracer, 2020-08-06:
@Kar-Mar hello, the PNG conversion is simpler:
1. make sure the PNG uses max level of compression == 6 (NOT HIGHER!)
2. use `bin2cc file.png file.c name_of_the_c_resource_variable`

Translations are much harder now, the build process will take only the PO files in near future, still **work in progress**. We were pushing the release fast. The generation of static data for localization (search structures and string tables) is done inside the translator unit test (which is not correct but was the shortest and safest path to achieve the goal)

DRracer, 2021-02-17:
@Kar-Mar in FW 4.3 the build process of languages takes only the PO files as promised. Is it an answer to your question? Can we close this issue?

Kar-Mar, 2021-02-21:
Thank you for the update info, I will look into the new version of FW

closed 2021-02-21 by Kar-Mar