prusa3d/Prusa-Firmware-Buddy #757: [BUG]Crashes when listing files containing å ä or ö (Swedish/German chars)

anderscallertun, 2020-08-06:
**Printer type** - MINI

**Printer firmware version** - 4.2.0-final

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
If the USB flash drive contains a file with a name with å, ä or ö (only tested those) the Mini crashes when listing those files when selecting "Print". The files are listed up to the failing character and then a second later you get a blue screen of death.
The same files were shown fine using FW 4.1.0.

**How to reproduce**
Put a file named "Servetthållare.gcode" on a USB flash drive. Insert the USB drive into the mini and select "Print".

**Expected behavior**
The files should be listed. (Possibly with a ? for the chars it can not display.)

**Video**
[video.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5031936/video.zip)

JohnnyDeer, 2020-08-17:
Hi @anderscallertun, diacritic is not supported by FW on MINI as well as on MK series. On 4.1.0 are files shown without letters with diacritic and are not printable. We will try to figure out how to avoid bluescreen on 4.2.0 but diacritic will not by supported at all. I´m closing this issue as Wontfix, thank you for understanding.

closed 2020-08-17 by JohnnyDeer

anderscallertun, 2020-08-17:
I'm not asking you to display these chars correctly. Displaying every char in the UTF-8 charset would be impossible.
Displaying a '?', a space or nothing at all instead of these chars is fine. I'm just asking you to make it stop crashing, since these files didn't make the mini crash with FW 4.1. So flagging it as "won't fix"  when you in fact want the same solution as I seems wrong...

JohnnyDeer, 2020-08-18:
@anderscallertun well then I agree and wontfix label is not right :)