prusa3d/Prusa-Firmware-Buddy #758: [BUG]<differences between Estimated end on Local Connect page and on display >

drakor007, 2020-08-06:
**Printer type** - [MINI]

**Printer firmware version**-  [ 4.2 ]

**Original or Custom firmware** [Original ]

**Optional upgrades** - [ Filament Runout Sensor ]

**USB drive**
 
**Describe the bug**
  There is a big difference in Estimated end on the display and on Prusa Connect Local page.
  Printer is connected to the Internet, ntp is not blocked.

[image]
[image]

JohnnyDeer, 2020-08-06:
Hi @drakor007 this issue is duplicate of #708. I´m closing this one.

closed 2020-08-06 by JohnnyDeer

drakor007, 2020-08-06:
> Hi @drakor007 this issue is duplicate of #708. I´m closing this one.

OK, I haven´t seen this issue on 4.1. Thank you.