prusa3d/Prusa-Firmware-Buddy #763: WATCHDOG RESET followed by crash dump(BSOD)

ve3mic, 2020-08-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.0

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
  WATCHDOG RESET (RED SCREEN) followed by crash dump (BSOD)

**How to reproduce**
  Happens shortly after auto bed leveling and purge line is printed

**Expected behavior**
  

**G-code**
  
**Crash dump file**
  see attached

**Video**
   not required
[image]

ve3mic, 2020-08-07:
[dumpdotbin.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5043329/dumpdotbin.txt)

Dump.bin

JohnnyDeer, 2020-08-11:
Hi @ve3mic, thank you for reporting.

JohnnyDeer, 2020-08-13:
We are constantly working on finding reason for this bug. Hopefully it will be fixed in next release. Thanks for understanding.
(Same for #771 & #791)

gameroptimizer, 2020-08-18:
Hi
My BSOD on prusa mini cause by Special marks on usb on one Gcode name.
language pack got messed up i took that out and my printer is working like dream again

DRracer, 2020-09-01:
@inferior-minion Does your error look like in issue #771? If yes, please try FW 4.2.1 and report back. Thank you.

JohnnyDeer, 2020-09-07:
If there isn't more BSOD cases I´ll close this issue soon. Please leave comment here if anyone will have BSOD on 4.2.1.

vintagepc, 2020-10-17:
I'll bite. Got a BSOD with the same message after a Watchdog reset on 4.2.1; this was the second or third layer of the print, if that matters. Will retrieve dump from USB after print finishes...

[image]

vintagepc, 2020-10-17:
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5396403/dump.bin.zip)

JohnnyDeer, 2020-10-19:
@vintagepc thank you for reporting, its new case.

DRracer, 2020-10-19:
@vintagepc Hmmm, this is really new ... tbh the second report like this on FW 4.2.1. You say it happened while printing? If it happens again, please try connecting the USB flash drive via a USB extension cable and/or use another USB flash.

vintagepc, 2020-10-19:
Correct - During the second or third layer (I'd watched the first one go down). 

Interesting you mention USB - After the reboot I went to restart the print and it choked up after selecting the print. 

I was able to browse the USB files, went to select the one in question and it froze for a few seconds and then displayed an empty directory listing. 

I checked the drive was securely inserted (it was) but I had to disconnect and reconnect it to get it to show again. After that I could select the file and the print completed without issue. I'd chalked that up to it being left in a bad state after the crash but it sounds like you think it might be a sketchy USB drive? (FWIW I'm using the stock one that came with the printer).

DRracer, 2020-10-19:
@vintagepc can you please do a fsck and may be also badblocks (nondestructive read/write) test of the USB flash drive on linux?

vintagepc, 2020-10-19:
Badblocks came back clean as well - as corroborated by the fact I was able to print the same file again without issue. 

```
# badblocks -nsv /dev/sdc1 
Checking for bad blocks in non-destructive read-write mode
From block 0 to 7864287
Checking for bad blocks (non-destructive read-write test)
Testing with random pattern: done                                                 
Pass completed, 0 bad blocks found. (0/0/0 errors)

```

marekmosna-prusa3d, 2020-10-20:
Hi @vintagepc,

can you please provide us your USB stick structure layout(folders, files, ...)? When you find out the way how to reproduce such an issue, that would be awesome and of course the most helpful. Basically, is before such a crash anything extraordinary during the process?

Thank you very much for your effort.

vintagepc, 2020-10-20:
Sure - nothing untoward; I simply added a few additional sliced files to the USB drive as received from the factory. 
[image]

So far it has not recurred but I will do my best to keep an eye out and note anything unusual surrounding the event.

marekmosna-prusa3d, 2020-10-22:
Hi @vintagepc,

Thank you again for your prompt answer. After some investigation I need to ask you another questions just to confirm what could happened:

1. Did the BSOD occur immediately after the printer startup? (< 20 seconds)
2. Last what you did was press the knob
3. Did you do anything else? (plug/unplug the USB stick, ethernet cable,...)

Thank you for your contribution.

JohnnyDeer, 2020-10-22:
I created a separate issue to make workaround more clear. All comments please write to #990, thank you.

closed 2020-10-22 by JohnnyDeer