prusa3d/Prusa-Firmware-Buddy #767: [FEATURE REQUEST]Signed firmware for Bondtech modifications

ogland, 2020-08-09:
Josef recently wrote this: "Recently when concern was raised about the appendix and firmwares for modded printers I offered to provide signed firmwares for well tested and safe modifications."

So now I'd like to cash in on this. Since the buddy firmware right after release of the Bondtech extruder and heatbreak removed the possibility to save esteps (and PID values?) this modification is somewhat crippled. As it is the most tested and safe modification that requires signed firmware, it should exist.

So a custom SIGNED firmware with esteps 415 is hereby requested.

As a sidenote I'd also like the heatsink cooling fan speed to not be limited, which it seems to be right now. Please at least unlock a secret menu where we can set it to full speed.

lennertvdkort, 2020-08-09:
Couldn't agree more! 
Yes i know you can modify the Gcode. But the filament change is also affected by the esteps.

enganear, 2020-08-09:
Yes, please. The Bondtech upgrade unlocks a lot of functionality for the Mini. Please release a signed firmware for it.

mobileagellc, 2020-08-09:
Yes, please!

bhbloom, 2020-08-09:
Yes please

LC3DPS, 2020-08-09:
Pretty please! 🙏 Thanks in advance Prusa

andrej-petrovic, 2020-08-09:
yes please !

andresfjs, 2020-08-09:
Yessss please.

philippbeckmann, 2020-08-09:
An even better solution would be to support advanced settings via a new dedicated M-code or the GUI.

derenma, 2020-08-09:
I _think_ I have seen a PID value save setting in the recent commits, and that is _probably_ going to hit in 4.3.0. _Maybe._ Absolutely don't quote me on that.

I totally agree with the Bondtech upgrade. The flexibility, strength and reliability is absolutely amazing. The stock extruder is perfectly fine, but it cannot handle filament that exceeds +/-0.03mm'ish. When doing draft prints, print quality absolutely does not matter and would rather burn through my awful spools in that regard. Tension adjustment of the stock extruder is, eh, not very good, but it works.

In Prusa's defense, EStep adjustment is a dangerous feature for new printer owners due to the abundance of bad information on the internet. Also, there should be no obligation for Prusa to compensate for out-of-spec filament. This is a company that deals in quality and precision, so why should they have to think about garbage filament?

BzuciakCZ, 2020-08-09:
+1 for it

No need of official Bondtech support, but return of 
M500 for ESTEPS would be great pain killer for Bondtech upgrade users.
Not to mention, lot of those users choose that upgrade as a last effort dealing with notorious heatcreep problems due to factory hotend assembly issues. 

Future collaboration with Bondtech on future development of MINI would be beneficial for both parties and mostly for customers themselves.

I like my MINI, it was a good printer (after pushing default heatbreak into heatsing) and is great printer after Bondtech upgrade.

ogland, 2020-08-09:
The only (official) reason Prusa removed M500 was that they were having issues with eeprom not working properly. It wasn't due to some UX ease thing. Esteps can be altered anyway. And even if M500 worked, you could still restore factory settings if some noob messed it up.

The mini doesn't have an e3d hotend unfortunately. It would be great if the stock heatbreak would work as good as e3d heatbreak.

JohnnyDeer, 2020-08-10:
Hi @ogland, thank you for suggestion. We are considering a solution to this issue.

cnEUL, 2020-08-17:
Any way, it would be very useful to have m500 support again. Please make savings to eeprom possible again. I had much problems with the original extruder and now i am using Bondtech. It works great. So also i would prefer to support such hacks.
But any way it would be also a good idea to heve better extruder / heatbreak from factory.
I love the mini, it is simple to use, small und fast. Also the GUI is very usefull. Come on...

josefprusa, 2020-08-18:
Hello everyone. That offer was in respect to the Bear project which is community project. With commercial products there will be some different rules. We will not sign anything which is not thoroughly tested in house for every release, which adds costs. This is all in the making but we will require the producer to have similar customer support to ours with some SLA in order to do it and some other things. 

Regardless the esteps and this issue, we will make similar hidden menu for the new platform as we are doing for the 8bit, which will solve this request.

ogland, 2020-08-18:
@josefprusa Fantastic! That sounds completely reasonable and very understandable. That reply is 11/10.

BzuciakCZ, 2020-08-18:
@josefprusa Understood, this is a good solution. Does it mean M500 will be still limited on Mini? (I need it currently only for ESTEPS anyway.)

PeterDHabermehl, 2020-08-18:
PLEASE please please let us enable M500 at least in custom build firmware. I totally understand your intention and I also can accept breaking the appendix. But then at least the fw should build with EEPROM_SETTINGS enabled. I would like to set E-Steps, Extruder loading/unloading max. speed, and PID values. I don't want to take you responsibility on what I do, so I would break the appendix. But the build has to work with EEPROM_SETTINGS enabled.
Thank you so much and kind regards
Peter

ogland, 2020-08-18:
@PeterDHabermehl You should read in the patch notes WHY they disabled it. It's not as easy as just enabling it.

bobc, 2020-09-01:
Looking at the code, Prusa already stores several values in EEPROM, adding new ones is quite easy. I don't understand Prusa's reluctance to maintain features that already exist in Marlin and people widely rely on.

ntmaster26, 2020-09-22:
Please please !! :)

dcs3473, 2020-09-25:
Yes Please...

d-zalewski, 2020-09-25:
New Mini owner with BondTech extruder and is upgrade. Please make this happen.

bergh-io, 2020-09-29:
Yes, we need this :) Any info about if, and if so when, this might me released?

JohnnyDeer, 2020-09-29:
Work on hidden menu for BondTech upgrade is in progress, will be in one of next releases.

Pcgrande87, 2020-11-29:
yes please !

sayeghr, 2020-11-29:
Is there any update on this issue? I’d really like to upgrade my extruder but am concerned the software doesn’t support it.

Pcgrande87, 2020-11-29:
@sayeghr yes I made a firmware,

sayeghr, 2020-11-29:
> @sayeghr yes I made a firmware,

What do you mean?

padigree, 2020-12-20:
Would be glad if this could implemented soon. It's a confusing to use different start g codes for every prusa I have, also it's difficult to sync this on every machine I slice with..

padigree, 2020-12-30:
Are there any news if it could be implemented in the next firmware?

Czarnyszarf, 2021-01-12:
Any updates? Commits about this feature?

stickybyte, 2021-01-12:
As I had seen in the source code, there seems to be a new menu item called „EEPROM“. Could be the menu @josefprusa talked about?
Maybe he can give more infos, if the feature will be part of the next release. I need it too...

Czarnyszarf, 2021-01-12:
Im more intrested in hotend PWM fan speed - it may be the solution for Bondtech heatbreak problems.

Also it would improve cooling on modified minis with v6 hotends , they run pretty hot due to inadequate cooling

s-taylor, 2021-01-15:
🙏 pretty please allow this. 3d printing is really a hobbiest discipline, and it's expected that people are going to tinker with their printers. Prusa knows this and supports this, they still provide customer support for modded printers so it's a logical step to support this. People are honestly more likely to generate bad prints and contact support because this feature is not baked in.

padigree, 2021-02-17:
2 1/2 more months are gone..

PeterDHabermehl, 2021-02-17:
Break the appendix...-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

SteeeveAT, 2021-03-17:
Just to push this up, and in hope Josef will realize this soon.

SHAXperiment, 2021-04-08:
Totally agree. Changing the E-Steps would be amazing. +1

jayseeee, 2021-04-23:
Any news on this M500 issue? I also want to be able to change the esteps and the PID values and store them into EEPROM.

BzuciakCZ, 2021-04-23:
#metoo Waiting so long to be able M500. No need special firmware for me. Just be able to store ESTEPS and PID.

Aquaerics, 2021-04-30:
we want the UPDATE now :-)

BzuciakCZ, 2021-05-15:
Hi guys. I just got an compromising idea. What about create support for post-start gcode, in which we can put our own initial values.
Something like AUTOEXEC.gcode :-)
Firmware and EEPROM will remain intact and if you mess something in that gcode, you simply remove USB stick and restart.

This might be useful not only for Bondtech modifications, but for many more
features bound to certain USB stick with distinct AUTOEXEC.gcode

I hate running my start gcode manually each time I want to load filament after printer start.

What do you think of this?

derenma, 2021-05-17:
@BzuciakCZ I like that idea in general! Even if M500 is never enabled (and I think I have seen the code changes for it, but not sure) your idea could be useful in a number of ways, for sure. Thumbs up!

Czarnyszarf, 2021-05-17:
Still no updates?  It's disappointing

BogdanH59, 2021-05-25:
I got my MINI+ about two weeks ago, which I've upgraded with Bondtech extruder & heatbreak during assembly. Reason being, I just wished MINI to give best possible results and to avoid unnecessary hassle in the future. But sadly, later I realized one can't upgrade MINI in a comfortable way, because it's impossible to store necessary G-code into EEPROM.
And even worse: because latest MINI's are delivered with modified board, it's also impossible to use custom firmware anymore. Not that I'm really interested in custom firmware by itself -but I could at least store G-code.
Yes, MINI works as expected, but so do many other cheaper 3D printers. Needless to say that I'm a bit disappointed in this regard.
I think this issue should be solved by Prusa as soon as possible.

derenma, 2021-05-25:
Well, you ain't wrong.

Prusa has been working on this issue, and I have been following most of the
code updates like a hawk.

I am speculating that most of the issues will be resolved after the next
board version.

Czarnyszarf, 2021-05-28:
> 
> Well, you ain't wrong. Prusa has been working on this issue, and I have been following most of the code updates like a hawk. I am speculating that most of the issues will be resolved after the next board version.
> […](#)

Where I can find info about that new board and what's happening ?

derenma, 2021-05-28:
You need to track the commits to the firmware, honestly.

BogdanH59, 2021-05-28:
Correction related to my previous post.
Explanation:
Per [Prusa documentation](https://www.help.prusa3d.com/en/article/flashing-custom-firmware-mini_14), appendix on board must be removed (yeah, I know) and jumper must be used. Because new board is missing one of three pins (and printer doesn't boot if jumper is used on remaining two pins), I assumed one can't load custom (non-signed) firmware anymore.
As it turned out, custom firmware can be loaded without using jumper. Prusa should mention that in documentation though.

pkg2000, 2021-05-28:
@BogdanH59 Prusa has updated their documentation with a note on this. It will see a larger overhaul to make it more future-proof in terms of hardware revisions.

matthewlloyd, 2021-06-28:
Anyone following this issue may be interested to know that I just released custom firmware that adds a menu option to configure e-steps, with presets for the Bondtech extruder with and without reversed wires, amongst other goodies:

http://github.com/matthewlloyd/Llama-Mini-Firmware#readme

enganear, 2021-06-28:
Thanks!!!

Stephen Chapman
Author - The Stonehenge Rift

ogland, 2021-06-28:
Cool @matthewlloyd

How did you fix the eeprom desync that made Prusa disable saving to eeprom in the first place?

matthewlloyd, 2021-06-28:
> How did you fix the eeprom desync that made Prusa disable saving to eeprom in the first place?

I couldn't find any specific details on the mysterious EEPROM desync (just [the patch notes](https://forum.prusaprinters.org/forum/general-discussion-announcements-and-releases/4-0-5-firmware-for-original-prusa-mini-2/) where they say Marlin EEPROM was causing "synchronization issues," and [this issue](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/364) where they say the values "were written inconsistently") so it's hard for me to say.

What I can say is that I didn't reenable Marlin EEPROM settings in Llama, I just followed Prusa's convention of moving the relevant settings into a proprietary section of the EEPROM. So I expect the llama has pronked right over whatever issues there were. :llama:

PippoAe, 2021-07-27:
I'd love to see a menu that allows to change settings if the bondtech extruder is installed.

JohnnyDeer, 2021-09-08:
Finally we released FW with promised hidden menu. We are sorry it took so long to develop this function. 
All you need to know is in the release notes. You can find new RC and try it here https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC.
Thank you all for patience.

ogland, 2021-09-08:
That's great! Do consider adding a feature to use the heatsink fan at 100% too, since many users are having issues with heatcreep.

BogdanH59, 2021-09-08:
Great news indeed! Even I already removed appendix (lost warranty), I still think added features might be useful for me one day.

I assume, many who upgrade the extruder also decide for (Bondtech) heatbreak upgrade. It's known that in this case different PID values are required. That means, to have exact PID values on startup, one still needs modified firmware (or run PID G-code after power on, or after factory reset).
Yes, we can run PID calibration which will be kept in memory (until reset)... But as Bondtech heatbreak has became a "standard" upgrade, it would be nice to have an **Heatbreak: Prusa / Bondtech** option in **HW Settings**. In my opinion this option would be more useful than **Extruder direction** (in case there's limited space on display).

anttikarhu, 2021-09-13:
Hoping to see support for the Bondtech heatbreak in the official firmware in the future :) 

Workarounds of modifying Prusa Slicer printer profiles or running some gcode after every boot both work, but are just workarounds. Using default profiles is easy to use and maintain, and running custom gcode is easy to forget :)

enganear, 2021-09-14:
Install the Llama firmware, it already has the ability to store PID values
in addition to esteps.

Stephen Chapman
Author - The Stonehenge Rift

[19 more replies]

closed 2024-07-04 by github-actions[bot]