prusa3d/Prusa-Firmware-Buddy #791: Mini firmware crash / error, "blue screen of death"

randomev, 2020-08-13:
21h print was almost done, about 1-2 hours left, left it print during night. In the morning I noticed blue screen on my Mini display.

Both printer fans were on. Heater was cold and bed was cold so safety was not compromized.

Firmware 4.2.0-final+2067. Just before print I had upgraded to this firmware. 
I have not experienced this kind of behaviour before.

My printer USB could be the reason, it seems that I need sometimes "wiggle" the USB-stick to get printer to recognize the USB-stick. Using USB-stick provided with the Mini.

Can provide GCODE of the print if needed (about 100 mb).

Picture of BOD attached.

[image]

Miyagi414, 2020-08-19:
I am getting the same error. First print after upgrading to 4.2.0 
 
Here is the gcode that I was running:
[CardBoxV3_0.2mm_PLA_MINI_9h13m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5097072/CardBoxV3_0.2mm_PLA_MINI_9h13m.gcode.zip)

[image]

MartinMajewski, 2020-08-20:
Right after the upgrade and during restart the above-mentioned crash.

Two hours into the first print with FW 4.2.0 and the same crash again.

[image]

What's going on and how to downgrade?

Best wishes,
Martin

MartinMajewski, 2020-08-24:
Another insight into that issue, and as @adynis has also said, the blue screen seems to be also (Micro-) USB related. After several successful long-time prints from the USB stick (after removing the aforementioned German Umlaut letter) I connected the printer to a RaspberryPi running OctoPrint. Tonight, a few hours into a print, the MINI crashed again.

I cannot recall it correctly, but I think I had the MINI also connected to my Mac, while the first bluescreens happened.

For the time being, I will revert to 4.1.0 to avoid losing print-results and time.

itsthejoker, 2020-08-30:
I have the same error that @MartinMajewski does -- a displayTask with a precise data access error. Photo here: https://twitter.com/filamentcolors/status/1299117013449797634 Worth noting that I do not have any non-ASCII characters on my USB drive, though I am still using the one that came with the printer. I have run the same gcode (and others) off of the same USB stick since the error with no problems.

rtyr, 2020-08-31:
New firmware version fixing random BSOD issues. https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.2.1

DRracer, 2020-09-01:
@MartinMajewski please note the issue with non-ASCII characters on the USB flash drive is still not fixed. This has never worked before and is still not supported. We'll fix the BSOD in this case to at least not crash the FW soon as well.

However, the BSOD randomly appearing during printing shall be fixed and hopefully never return - that applies to reports from @Miyagi414 and @adynis .

brentday, 2020-09-01:
Hello 

What does it mean when the printer stops printing and goes to a white screen. I'm using the USB stick that came with Mini.

JohnnyDeer, 2020-09-02:
Hi @brentday, white screen mean issue with LCD communication. Please check if your connectors are properly connected.

MartinMajewski, 2020-09-02:
> @MartinMajewski please note the issue with non-ASCII characters on the USB flash drive is still not fixed. This has never worked before and is still not supported. We'll fix the BSOD in this case to at least not crash the FW soon as well.

Thanks for the information! Usually, I don't use non-ASCII characters, this was a typo from my side.
I will raise a feature request at the PrusaSlicer repository because I think it would be a good addition to prevent those typos beforehand.

PR #872 (2020-09-07): Fix BSOD with non-ASCII chars in file names
BFW-1236

From now on we even may be able to print the files with filenames containing non-ASCII characters, but please note you shouldn't abuse it, there are still some edge cases, when it may break the list of files. It shall not crash hard anymore though.

Fixes #791
closed 2020-09-07 by DRracer