prusa3d/Prusa-Firmware-Buddy #798: PID autotune failed! - Temperature too high

kgolger, 2020-08-16:
Printer type - MINI

Printer firmware version -  4.2.0

Original or Custom firmware

USB/Octoprint V1.4.2 octopi 1.17.0

**Describe the bug**
When trying to perform the bed autotuning via the Octoprint terminal with the command "M303 E-1 S60 C5", I always get the following error message:

PID autotune failed! - Temperature too high

No matter what temperature settings I use, the error remains.

The nozzle autotuning works fine with "M303 E0 S170 C5".

In addition the values can be stored with M500, but after switching the printer off/on the original values are up to date again.

**How to reproduce**
Start Octoprint terminal or Pronterface and type "M303 E-1 S60 C5"

**Expected behavior**
Please enable bed/nozzle autotuning with subsequent permanent saving to the Eeprom, so that the values are available again even after switching the printer off/on. These are technical settings which belong to the printer and should not be anchored in the slicer and if the settings are already available, they should work properly.

**G-code**
No gcode

**Crash dump file**
No crash

**Pictures**

[image]

[image]

JohnnyDeer, 2020-08-17:
Hi @kgolger, Marlin PID autotune results were worse than the manual setting, so we redesigned whole temperature control. Therefore Marlin PID autotune is not supported by fw at all. If you have encountered any possibility of improvements, we would like to examine them, thank you.

tonylee1918, 2021-02-07:
> Hi @kgolger, Marlin PID autotune results were worse than the manual setting, so we redesigned whole temperature control. Therefore Marlin PID autotune is not supported by fw at all. If you have encountered any possibility of improvements, we would like to examine them, thank you.

Hi JohnyDeer, 

Does this apply to the PID Autotune for the hotend as well? I ask bcause the original question was for the bed if I understand it. Thanks,

kgolger, 2021-02-07:
PID autotune for the (Mini) hotend works for me.
wbr,
Karl

elpeter25, 2021-02-19:
Is the heatbed PID autotune still not supported? I'm receiving same error "PID autotune failed! - Temperature too high" and I really would like to give it a chance as I'm experiencing issues like below:

https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/why-you-should-run-pid-tuning-for-heatbed/

Kr

matthewlloyd, 2021-07-04:
This bug was introduced by this commit: https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/f093445e1117da2010db4c323e5549f030731a4a

The value of `H_BED` was inadvertently changed by this commit from -1 to -2.

The workaround is to use E-2 instead of E-1 to PID autotune the bed:

`M303 E-2 S60 U1`