prusa3d/Prusa-Firmware-Buddy #799: [FEATURE REQUEST] Display cooling fan speed value in percent, rather than an unsigned byte.

derenma, 2020-08-16:
**Describe the expected functionality**
This is really picky and not a huge issue.

I believe that it is more intuitive for new users to change cooling fan speed in a percentage rather than the stock Marlin value of between 0 and 255.

Sure, there is a rounding problem involved with this, but that should be easy to solve for.

JustAnother1, 2020-08-20:
I'm curious what would be the best way to solve the "rounding issue" ?

I can think of these options:

1) Have 0..100%(changes in 1% steps) so only 100 steps with each percent mapped to one of the 256 steps available.  So basically loosing 156 Steps resolution.

2) Have 0.0 .. 100.0% (changes in 0.1% steps) so 1000 steps but many steps map to the same value. So changing the setting by 0.1% will in about 3 of 4 times not change the fan at all.

3) Calculate the percentages for the 256 steps and have the last digit (the 0.something) jump to the next available number.

derenma, 2020-08-23:
> Have 0.0 .. 100.0% (changes in 0.1% steps) so 100 steps but many steps map to the same value. So changing the setting by 0.1% will in about 3 of 4 times not change the fan at all.

I like this idea the most, FWIW. There is little precision lost and it's much more intuitive.  The precision that could be lost is actually irrelevant to something as brute force as cooling fan speed.

closed 2024-07-04 by github-actions[bot]

JustAnother1, 2024-07-04:
What a nice FU.

Prusa-Support, 2024-10-12:
I'm sorry to see that the issue went stale without interventions after the initial notice but this issue can actually assigned as completed.
The print fan speed can be set in `%` value now, in accordance with the parameters in Prusa Slicer, so rounding shouldn't be a concern.

Michele Moramarco
Prusa Research