prusa3d/Prusa-Firmware-Buddy #810: [BUG] Manual MBL crashes the nozzle into the bed and terminates prematurely. (Video link included.)

derenma, 2020-08-17:
**Printer type** - [MINI]

**Printer firmware version**-  4.1.0 Final

**Original or Custom firmware** Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint** - USB/Octoprint

**Describe the bug**
Manual MBL crashes the nozzle into the bed and terminates prematurely.

**How to reproduce**
Reboot Mini. Calibration -> Mesh Bed Leveling

**Expected behavior**
My printer becomes sentient, maybe?

**Video**
https://youtu.be/AD7ooszM_KU

JohnnyDeer, 2020-08-18:
Hi @derenma, your MINDA cable is probably broken, please contact our Support, they will help you.  (https://shop.prusa3d.com/en/ - chat icon in lower right corner). I´m closing this issue as Invalid.

derenma, 2020-08-18:
The MINDA cable seems perfectly fine. You can see the printer detecting the bed in the first two probes, but it fails after. This issue only happens with manual MBL. The failure will not happen when I run normal prints: https://youtu.be/0pnC8OScJ8U

JohnnyDeer, 2020-08-19:
Well now i did reproduce it and this is FW bug. I´m reopening issue and we will fix it for some of next releases. Thank you for reporting.

vintagepc, 2020-10-21:
FWIW this can also happen if you send a G29 without a G28 first, since G29 alone does not re-home, only travels to what the printer thinks is 0,0

I observed similar after the steppers were disabled and the axis bumped -  a subsequent G29 assumed the printer was still at 0,0 and proceeded to travel off the right side of the bed and crash because the MINDA was off the edge of the sheet.

vintagepc, 2020-10-23:
Just had this happen again; this time with a heated nozzle so there's now a nice divot in my sheet. Hopefully it can get addressed in the upcoming release, even if it's just as crude as enforcing a G28 all the time with that menu option.

derenma, 2020-12-25:
@JohnnyDeer 4.3.0 has resolved this issue. Woohoo!

JohnnyDeer, 2021-02-10:
> @JohnnyDeer 4.3.0 has resolved this issue. Woohoo!

Then I'm closing as solved, thank you.

derenma, 2021-07-28:
@JohnnyDeer We may need to reopen this issue as it seems related to the original problem.

I am now running 4.3.2-RC2 (Upgraded to Mini+) and just noticed that this issue will pop back up **_after_** a successful MBL is ran manually. (I am not sure if the issue is a result of 4.3.2-RC2 or if the problem still existed in 4.3.0 as well.)

Replication Steps:

1. Reboot Printer
2. Goto: Calibration -> Run MBL
3. Allow MBL to complete.
4. Goto Settings -> Disable steppers.
5. Manually move hotend to center of bed
6. Rerun: Calibration -> Run MBL

Results on 2nd MBL Run:
1. Y-Axis Homes 
2. X-Axis _doesn't_ home
3. Hotend travels to X-Max
4. Hotend Z-Axis plows into bed. Eeek!
5. Reboot printer
6. One MBL will complete successfully after reboot.

derenma, 2021-07-28:
@vintagepc I just re-read your comments and they align with what I just posted above, me thinks.

vintagepc, 2021-07-28:
Yes, this appears to be an inherent issue with the printer assuming it is still where it last was. I can't think of any reason not to always enforce a G28 first prior to MBL, TBH

vintagepc, 2022-01-08:
This bug is back in 4.3.4. Just had it happen, "mesh bed levelling" should just always auto-force a home before running.

prochazkaml, 2022-01-09:
I've had a similar experience. When I turned on the printer and selected MBL from the Calibration menu, the printer moved the Y axis to 0, didn't bother moving the X axis at all (it stayed in the middle) and the nozzle went straight into the bed.

Of course, bed mesh leveling (after auto-home) in gcode works perfectly fine.

derenma, 2022-01-10:
@JohnnyDeer This is still very much an issue. The printer is "assuming" the hotend hasn't changed position after it's last MBL which still causes a crash.

I'll recreate this issue with a video in the AM.

derenma, 2022-01-10:
In theory, if I was able to get access to an independent print farm of Mini's, we can theoretically kill them all here. This is kinda serious.

derenma, 2022-01-10:
My first post was Aug 17, 2020. The fair disclosure period is in full force here.

vintagepc, 2022-01-10:
I think that's a bit of an overreaction. There are any number of g-code commands you could send to an unsecured printer to potentially cause it to damage itself... this is not very different from (say) commanding it to heat up, zero, and extrude a large quantity of filament and blob itself.

JohnnyDeer, 2022-01-11:
I can confirm that this bug appears on the latest FW version again. @derenma thank you for reporting.

derenma, 2022-01-19:
@JohnnyDeer I go in speculation rabbit holes, so excuse that.

I have been out of town for the last week with no internets, so I am chomping at the bit to look at the MBL code. :)

derenma, 2022-01-19:
> I think that's a bit of an overreaction. There are any number of g-code commands you could send to an unsecured printer to potentially cause it to damage itself... this is not very different from (say) commanding it to heat up, zero, and extrude a large quantity of filament and blob itself.

You bring up a very good point. I am curious to see what happens if I issue an MBL, print a part and re-issue an MBL again. That wouldn't be nefarious code.

danopernis, 2024-05-03:
@derenma can you confirm with the latest 6.0.0 firmware? Since similar issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1268 was closed perhaps this can be closed too?

closed 2024-07-11 by github-actions[bot]