prusa3d/Prusa-Firmware-Buddy #826: [BUG] 4.2.0 - First Layer Calibration only heats for PLA

MartinMajewski, 2020-08-21:
- Original Prusa MINI
- FW 4.2.0
- Official Release
- Filament Outrun Sensor
- USB Stick

After having the blue screen of death issue
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/791

I tried it with a factory reset to see if this would help.

Selftest went well, but I discovered an issue with the first layer calibration.

As shown in the video, after starting the calibration process, you get the choice to select a material type. After selecting, the printer begins to heat up to the correct temperatures. However, to get to the actual calibration process, you have to click on next multiple times. As soon as the calibration screen appears, the temperatures set to PLA standards (215/60). These temps are too cold for filaments like, e.g., PETG (and please, only because Jo likes PLA, it does not mean everyone has an open PLA spool lying around in case of a calibration event! 😉).

**Video**

https://youtu.be/1yt3OHDIIik

JohnnyDeer, 2020-09-10:
Hi @MartinMajewski I can confirm this bug. Thank you for reporting.

JohnnyDeer, 2021-02-10:
I tested this issue on 4.3.0 and works well. @MartinMajewski could you please try it too and update issue?

MartinMajewski, 2021-02-10:
Sure, I will test it in the evening and report back!

MartinMajewski, 2021-02-16:
The first layer calibration works flawlessly now.
I tested it in between prints and after a factory reset.

Btw.: Wizzard and 1. Layer Calibration are much better now - and finally printing the pattern from front to back! :-)

Thank you, issue is resolved!

JohnnyDeer, 2021-02-16:
@MartinMajewski thank you for update, I´m closing issue as solved :)

closed 2021-02-16 by JohnnyDeer