prusa3d/Prusa-Firmware-Buddy #832: [ENHANCEMENT] First Layer Calibration improvement

mneuhaus, 2020-08-23:
Hey,

i love my Mini and especially the way it helps me fine tune it. One thing kind of bothers me though, the First Layer Calibration is a good start, but it is kind of "hit and miss" where you have to repeat it to get the best result.
My idea is, that instead of the one square at the end there could be a row of ~15 Squares each 1x1cm that slightly differ in Z-Height. So, in the middle the current Z-Height Level, to the left it would decrease in small steps, to the right it would increase in small steps.
This would make it easy to "eyeball" the while zig-zagging and then letting it finnish. After the row of squares are done you can check with the finger and do a final z-height adjustment based on the best square. So, if for example the third square from the center is the best and the "difference" of the square is 0,005 you would decrease the final z-height by 0,005*3.

You could even add the Question to the UI like this: "Which square feels the smoothest: [6]" and do the adjustment in the Software based on that selection

MartinMajewski, 2020-08-24:
I am asking for a better 1. layer calibration pattern for a long time now. I don't know who really thinks that the current one is a good choice, but this person seems not to use the calibration process very often! 🤨

Two things I would change to make the process better:

1. Just print a large filled square! You can judge the adhesion and correct distance between the nozzle and bed far better in an area than a single line. It mirrors the real deal far better than a single line. Being too far up is obvious in both cases, but being too close to the print surface is much better visible on lines that are side by side to each other, as in a rectilinear infill pattern. Moreover, you can adjust the height back and forth to verify that you hit a sweet spot because you have more time and more "data" to look at. Using black filament on a black textured surface is so hard to judge when only a single line is present. I don't want to change filament just to get the 1. layer calibration done. If you're fine with the result, there should be a "finish" option to confirm and get out of the process gracefully. No need to wast time and filament when the results are already perfect. So nobody can convince me that a square would take too much time and material to complete!

2. Print the pattern from the front to the back of the build surface. Who came up with the idea to cover the results with the extruder, anyway? Printing from back to front, like it is the case now, is only good for post-print analysis but this process is called "live adjust". I want to see the effect of my changes immediatelly and be able to try out the stickiness of the filament to the print surface with my finger, without the extruder getting into my way. The results should be presented immediately, not hidden!

Moreover, there are still bugs in the filament temperature options for the calibration process, as I have shown  here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/826

mneuhaus, 2020-08-24:
i'd be willing to code it myself, but right now i'm a bit hesitant because of the warranty voiding

padillla, 2020-09-26:
You don't break the warranty if you test a regular live z with the gcode of the pattern you rather have loaded on the USB, I personally think 4 squares within with a spiral will help better

DRKMSTR, 2020-11-01:
This is similar to the issue / enhancement recommendation of #309

closed 2024-07-04 by github-actions[bot]