prusa3d/Prusa-Firmware-Buddy #836: [BUG] can't move axes between first layer calibrations

hainjedaf, 2020-08-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI] **MINI**

**Printer firmware version**-  [e.g. 4.0.5, ...] 4.2.0

**Original or Custom firmware** [Original/Custom] **ORIGINAL**

**Optional upgrades** - [e.g. Filament Runout Sensor] **Filament Sensor/textured bed/smooth bed**

**USB drive or USB/Octoprint** **BOTH**

**Describe the bug**

When you do a first layer calibration, the printer moves to the right front position of the bed. Z=6 or thereabouts.
Then the LCD asks if you'd like to repeat the procedure.
If you say yes, the LCD advises you to clean the bed.

**HOW?**
The print head and Z axis remain in the way!

Manually you can move X and Y but Z is almost impossible.

**How to reproduce**
Do a 'first layer calibration'
Answer the question about repeating the procedure wit 'yes'.

**Expected behavior**

Either: 
- a small menu to move the axes should appear
or
- the printer raises Z and moves table forward to enable cleaning.
then
- option to continue to redo the 'first layer calibration

**G-code**
no G code involved AFAIK

**Crash dump file**
  no crash, no dump

**Video**

derenma, 2020-08-27:
Disabling the steppers between calibrations would be a solution as well.

JohnnyDeer, 2021-02-10:
@hainjedaf the first layer calibration procedure is changed in 4.3.0, now you can check and clean bed without needing to move axis. I´m closing this issue as solved.

closed 2021-02-10 by JohnnyDeer