prusa3d/Prusa-Firmware-Buddy #839: [FEATURE REQUEST] Visualisation of mesh bed leveling

obi134, 2020-08-26:
**Printer type** - MINI

It would be great if you could visualize the measured z-offsets from mesh bed leveling. You could call e.g. the manual mesh bed leveling from time to time and have a look on the measured offsets and can see if the bed gets waved or misaligned.
I imagine the values shown as z-offsets in 4x4 grid and maybe colored like a heatmap

bbbenji, 2020-08-26:
Don't think this will be added anytime soon. In the mean time you can use this: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/wiki/Firmware-Specific-GCODE-Examples#prusa-mini

derenma, 2020-08-26:
@obi134 like @bbbenji pointed out, it's available with OctoPrint.

However, you are not going to like the results of the mesh when you see them, I assure you. Since there is no way to actually manually adjust leveling without doing a complete disassembly of the bed, you are going to risk damage to the printer unnecessarily. Having just replaced my bed thermistor, I can tell you this is not a fun process if you do not want to disturb the belt assembly in any way.

Additionally, you will have to work with 9 points of leveling and not the typical 4 points that are found in cheaper printers.

My bed is tilted between -0.2mm to 0.1mm to in one direction (with a bend) on my Mini. That is very significant, but it's nothing that the mesh leveling can't handle. The precision side of me is cringing but the realistic side of me not too worried since I still get good prints.

bbbenji, 2020-11-20:
You can now visualize your bed here: https://bbbenji.github.io/PMSBLM/

However, you still need to run`G29` from some console.

closed 2024-07-04 by github-actions[bot]