prusa3d/Prusa-Firmware-Buddy #848: [BUG]<Enter comprehensive title>

hainjedaf, 2020-08-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [e.g. 4.0.5, ...] 4.2.0

**Original or Custom firmware** [Original/Custom] ORIGINAL

**Optional upgrades** - [e.g. Filament Runout Sensor] Filament Sensor 

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint BOTH

**Describe the bug**

values changed using M92 are not stored in NVRAM when issuing M500

Example:

S> M92 E334.54
R> OK
S> M92
R> M92 X... Y... Z... E334.54
S> M500
R>
.....
M92 X.... Y... Z... E 334.54
....
S> M501
R>
.......
M92 X.... Y.... Z.... E325
.....
S> M92
R: <M92 x.... y.... z.... E325

So upon M501, Reset or power off, M92 falls back to default. User update not stored.

**How to reproduce**
Change M92 Exxx
read back M92
check Exxx value
issue M500
check M92 Exxx value

The either:

issue M501
check M92 Exxx value

or

reboot/powecycle printer
check M92 Exxx value

or

do any calibration from LCD menu
check M92 Exxx value

**Expected behavior**

M500 should permanently update EEPROM stored values.
After a reboot, powercycle or calibration, adjusted values should be retained.
Example

M92 Exxx
M500 
reboot
M92
(Exxx value should be same as set before, not 325 as hardcoded in firmware.)

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

M92 E334,54
M500
M501

\**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

No crash dump availeble
**Video**
  Please attach a video. It usually helps to solve the problem.

hainjedaf, 2020-08-31:
Version 4.2.1:

M500 no longer reports, just answers [OK].

M501 resets values to stored defaults (i.e. deletes user changes)

derenma, 2020-08-31:
M500 was removed by design.

Check out this thread where we talk about M500 & signed firmware.

JohnnyDeer, 2020-09-01:
Hi, this issue is duplicate of #767. I´m closing this one.

closed 2020-09-01 by JohnnyDeer