prusa3d/Prusa-Firmware-Buddy #849: [QUESTION]<Is There a Way to Disable Live Z-Height Viewing?>

SiberianSilverFox, 2020-08-31:
In the versions of the firmware before 4.2.0 The layer height that was displayed on the screen showed the value of what the slicer said the layer would be not the live height value. This was great for me as I would be able to have the slicer open and see exactly what layer my print was on. Now with the updates to the firmware it shows the live value, and I am no longer able to see what layer I am on because it changes as the nozzle moves around.

Is there any way that I can make it so I can see what layer I am on? While the live layer height is great to see how my bed is leveled I really miss being able to follow how my print is going more precisely in the slicer.

x10102, 2020-08-31:
Probably a duplicate of #608

JohnnyDeer, 2020-09-01:
Hi, as @x10102 wrote this is duplicate. I'm closing this issue.

closed 2020-09-01 by JohnnyDeer