prusa3d/Prusa-Firmware-Buddy #850: [BUG]M851 in filament start GCODE ignored when starting a print

bmangodt, 2020-09-01:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.0

**Original or Custom firmware** Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
 USB drive

**Describe the bug**
  I have different filament profiles to account for the change from the powdercoated steel sheet to the smooth PEI steel sheet. To make this progress easier I added a M851 gcode to the start gcode in that filament profile in Prusaslicer. When I start a print changing plates the M851 executes and I can see the offset in the tune menu, but it still prints either too high or too low. Only stopping the print and restarting it, basically printing again with the offset already set, it functions correctly. Could it be the filament start gcode executes after the printer start gcode so the printer homes and levels the bed with the old offset instead of the new? Does that have any effect?

**How to reproduce**
  Change steel sheet and set offset for the new steel sheet in a filament profile in Prusaslicer.

**Expected behavior**
The printer takes into account the new offset.

**G-code**
 Add M851 to the end of the start gcode of a filament profile in Prusaslicer. My values won't work on other printers.

JohnnyDeer, 2020-09-08:
Hi @bmangodt in one of future releases we will implement feature to choose between plates as on i3 series and that will avoid this kind of issues. M851 is not supported or rather tuned and dont have priority to be. Thank you for understanding.

JohnnyDeer, 2021-05-12:
Feature with choosing between plates is already implemented so is not necessary to use M851which is not supported by MINI FW.  I´m closing this issue.

closed 2021-05-12 by JohnnyDeer