prusa3d/Prusa-Firmware-Buddy #856: Crash of Version 4.2.1 BSOD

chemiefrik, 2020-09-01:
BSOD when trying to print from USB Drive.
I added the Crash-Dump by changing the name name of Dump.bin to Dump.zip to fool Github. Just rename to dump.bin
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5158483/dump.zip)

**Printer type** - [MINI]
**Printer firmware version**- 4.2.1
**Original or Custom firmware** [Original firmware] everything as bought
**Optional upgrades** -  Filament Sensor Yes
Print via USB Drive
**Describe the bug**
  BSOD when clicking on "printing" in the Home menu. When listing folders some unreadable appears.
Could be reproduced three times in a row. As I am german and my system is on German. My thought is that a foldername with ä,ü,ö is causing this?

After changing the folder-name from" Sprühdose" to "Spruhdose" the instant BSOD went away.

**How to reproduce**
Boot system. Click in menu on printing see some folder names and after 2 or 3 seconds BSOD

**Expected behavior**
That I can read folder names to select G-CODE. Using the Provided USB Stick a compatibility issue shouldnt be the issue.

**G-code**
Did not reach. G-Code relevant areas.

**Crash dump file**
Attached as Dump.zip change to dump.bin

anderscallertun, 2020-09-02:
This looks like a duplicate of #757 . That one was closed without being fixed.

JohnnyDeer, 2020-09-02:
Hi @chemiefrik this is known bug and don't have high priority as BSOD while print. Fix is planned.

DRKMSTR, 2020-10-13:
I have had this problem on 4.2.0, ethernet was not connected. 

Printer interface was becoming sluggish and unresponsive (memory leak maybe?) in the hours before the problem occurred.

JohnnyDeer, 2021-02-10:
This bug is fixed in 4.3.0, the printer will not crash. File is shown and unable to print as before. I´m closing this issue as solved.

closed 2021-02-10 by JohnnyDeer