prusa3d/Prusa-Firmware-Buddy #864: [BUG] Print head homing after print (with Video)

derenma, 2020-09-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, Bondtech heatbreak/extruder

**USB drive or USB/Octoprint** - USB

**Describe the bug**
Printer initiates home after print.

**How to reproduce**
Unknown. Attaching video, dump.bin and gcode.

**Expected behavior**
The printer does not attempt to self-destruct.

**G-code**
Attached.

**Crash dump file**
Attached.

**Video**
 This is a reproduced error that I was able to get on video the second time: https://youtu.be/E0-OvU_h4do

**Theory/Debugging**
I did one manual homing during this power cycle. I will reboot the printer and update this post if there are any changes in the next test print.

**gcode and dump.bin**
[gcode_and_dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5172747/gcode_and_dump.zip)

**gcode modifiers in slicer**
M92 E415
M301 P13.54 I0.98 D46.58
M500

neophyl, 2020-09-04:
#733 ?

derenma, 2020-09-04:
@neophyl Possibly. I am not getting the purge sequence, just the home.

Doing a few more tests today to see if I can isolate the issue. This seems like it would be a pain for Prusa to replicate without me sending my whole machine back to them. ;)

I am really thinking it has to do with the M301 that I just added since that is the only notable change that I have done in the last day. There are a few scenarios I can still test and will reconnect my OctoPrint pi to see if the console will tell me anything.

derenma, 2020-09-10:
@JohnnyDeer This is a difficult one to replicate, so godspeed.  If you want the full "good" and "bad" gcode, let me know. (I unfortunately had Prusament Orange loaded printing 20x test cubes, so that sucked. ;) )

JohnnyDeer, 2020-09-10:
Hi @derenma yes please, it would be helpful. We think this issue may have same reason as #733.

derenma, 2020-09-10:
@JohnnyDeer Attached the two gcode files at the end of the problem description. Once I was confident that I found the problem, I changed the test cube to around 5mm^3 and it had no effect on the underlying problem. These super-quick prints were printed in Prusament PETG.

derenma, 2020-09-11:
The error has to be in here somewhere: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/e2445b148f679a893ddc33f1d6f773d1e07c115f/src/common/gcode_file.cpp#L94-L124

'=' is still getting processed after the comments. Totally cool, as I have no idea what ya'll are doing with the comments. I am thinking there is an exception issue, which is total speculation.

derenma, 2020-09-12:
@JohnnyDeer I am curious if you are able to replicate. Please, let me know.

derenma, 2020-09-12:
If not, it points to a hardware fault. It you can do it, it's firmware.

danielo515, 2020-09-15:
I am also having a similar behavior

zoltan-l, 2020-10-05:
derenma, did you try to remove the M92 E415 G-code from your .gcode file?

derenma, 2020-10-06:
@zoltan-l I did not. I have to admit that is a good troubleshooting step that I overlooked, but setting esteps for the Bondtech extruder is super critical, IMHO. I'll try this tomorrow in the morning.

zoltan-l, 2020-10-07:
@derenma 
thanks, looking forward to your response

derenma, 2020-10-07:
@zoltan-l It's taking me a bit to revert back to a broken state. :) Should be there soon.

zoltan-l, 2020-10-07:
I miss any video proving all have the same issue. I could reproduce the issue using (almost ) identical G-code with and without M92

derenma, 2020-10-07:
@zoltan-l Tried the following:

1. Commenting out the M92 at the top of the code - Printer failed.
2. Deleting the M92 completely from the top of the code - Printer failed.
3. Attempted a NOP slide into `; start_gcode = AAAAAA...` - Print worked as expected. (I have no idea what the stack looks like for this control board, or if I could even attempt such a hack.  That was more just for lulz.)

derenma, 2020-10-07:
Since deleting two lines of gcode above `; start_gcode =` had no effect so the problem is isolated, I believe, to how the comments are getting parsed at the end of the gcode file on the printer itself.

I have posted and deleted a few wild theories in this thread about this, but alas, they were not constructive in hindsight.

derenma, 2020-10-08:
@zoltan-l Oh S***. I went to start a print just now and the NOP slide actually worked somehow! I thought the last print test ended in a "complete" state, but the printer actually paused after the print. It didn't actually complete!

This opens some doors for exploration, that's for sure.

derenma, 2020-12-20:
[ deleted crazy engineer ranting here ]

zoltan-l, 2021-01-12:
The cause is the same like https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/733
workaround same - remove from Gcode the long ( longer than 65 characters) comment line(s) which is (are) misinterpreted by parser.

derenma, 2021-02-11:
@zoltan-l Wheeeeeee! I hope I was able to provide enough breadcrumbs for you to work with. Cheers!

zoltan-l, 2021-02-11:
@derenma 
Matt thanks, you were great !

derenma, 2021-05-16:
Woohoo!  Just saw the recent commit to fix this issue and it looks solid, FWIW.

derenma, 2021-05-18:
LGTM. Ship it.

derenma, 2021-05-18:
(not a fan of statically defining issues, but that is filtering 101.)

JohnnyDeer, 2021-05-25:
New FW version with fix of this bug is out. @derenma please try it out ad let know if this issue can be closed :)
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2-RC1

derenma, 2021-05-26:
@JohnnyDeer @zoltan-l BAM! Put a bow on it and ship it. The "bad" g-codeI posted a few months ago prints just fine under the exact same conditions.

For lulz, I just used a garbage USB drive and a USB extension that is possibly going bad. I had suspected there were additional USB issues that _might_ have been part of this problem, but wasn't sure if there was an association or not. I think I deleted some speculative rants about that in the past.....

JohnnyDeer, 2021-06-01:
@derenma thank you for all your help we appreciate your approach :) 
I´m closing this issue as solved.

closed 2021-06-01 by JohnnyDeer

derenma, 2021-06-01:
Yay!