prusa3d/Prusa-Firmware-Buddy #866: [BUG]<ORIGINAL PRUSA MINI - Blue screen - TASK: webServerTask> #2821

Psychorages, 2020-09-04:
Printer type - ORIGINAL PRUSA MINI
Printer firmware version- 4.2.0-final+2067

Describe the bug
not sure, but i may have try to access to Prusa connect at this moment of print
Blue screen @ 74% print job :
[image]

To Reproduce
print with lan on

Expected behavior
no blue screen :')

G-code
[Base_0.2mm_ABS_MINI_6h46m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5176854/Base_0.2mm_ABS_MINI_6h46m.gcode.zip)

zoltan-l, 2020-09-05:
Hi ,it is solved by 4.2.1 including the root cause ,should you find it again, let us know please.

JohnnyDeer, 2020-09-07:
Hi @Psychorages as my colleague wrote this is fixed in 4.2.1 also it´s duplicate of #763 #771 #791. I´m closing this issue as duplicate.

closed 2020-09-07 by JohnnyDeer