prusa3d/Prusa-Firmware-Buddy #868: [ENHANCEMENT] Add Mesh bed leveling temperatures as option into Preheat menu

mojeto, 2020-09-06:
**Printer type** - MINI
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
I print often with filament pre-loaded in printer from previous print. I like to preheat my printer before first print, but all preheat options are way over 170°C for nozzle. As a result, I have to wait for printer to cool down the nozzle to start probing before print starts on preheated printer. 

**Describe the expected functionality**
I want to be able to set printer to preheat nozzle to 170°C and bed to 60°C. So printer can preheat and I can start print faster still.

jkavalik, 2020-09-06:
That already works since I think 4.1 - the preheat menu shows for example PLA 215/60 but it actually preheats only to 170/60 and waits there for print start. Similar with filament load - it loads at the right temperature but after confirming it starts cooling down to 170 immediately

mojeto, 2020-09-06:
You right, I installed 4.2.1 and it works as you described. That's great. So, I don't need special option in preheat menu. Thank you.

JohnnyDeer, 2020-09-07:
Hi @mojeto as @jkavalik wrote, this function is already implemented in the FW. I´m closing this issue as solved.

closed 2020-09-07 by JohnnyDeer