prusa3d/Prusa-Firmware-Buddy #869: [ENHANCEMENT] Move tool higher after low print finished

mojeto, 2020-09-06:
**Printer type** - MINI
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
Printer moves X and Y axes to max position on print finish, but doesn't move Z axis. 
This cause few problems for low prints
* It's harder to remove bed sheet from under nozzle.
* It's hard to remove extra filament coming from cooling nozzle.
* It's harder to return bed sheet under low positioned nozzle.

**Describe the expected functionality**
I want printer to detect low prints and move Z axis up to at least 35-40 above bed on print finish.

Related to #242 and #637

neophyl, 2020-09-08:
The mini (like the MK3) has a line in the Printer Settings > End Gcode 
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+5, max_print_height)}{endif} F720 ; Move print head up
This line moves the print head up 5mm after a print as long as it doesnt exceed the max print height.  The Mk3 profile moves the head up 30
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} F720 ; Move print head up.

Not sure why the mini is set to only move up 5mm but if you want it higher just change the value (layer_z+??).  No need for any change to printer firmware at all.

mojeto, 2020-09-08:
I'm new to this.
Is this End Gcode a PrusaSlicer setting that can be changed ? @neophyl

neophyl, 2020-09-09:
Its not a simple setting as such, however it can be changed.
Open up Prusa Slicer (I am making the assumption that you are using PS).  Make sure you are in advanced mode or the options wont be available to you.  If you are using a different slicer then it should have similar sections to define start/end gcode.

Go to printer settings and you should see a section called 'Custom G-code'.  In that section are the areas for the Start, End and before/after layer change codes.  Any g-code in the start area is performed before the printer starts your actual print, so all the preheat, homing , mesh levelling etc is set up here.  The End g-code is performed after the print has finished and does things like turn the heaters off and moves the bed forward etc.

In the End G-code section you should see the line I posted above.
_{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+5, max_print_height)}{endif} F720 ; Move print head up_

This line is telling the printer to move up 5mm unless the printer is already at the maximum height.  This is basically to stop it trying to drive the extruder higher than it can go if you have done a tall print.  On the Mk3 printers profiles there is the same line but its set for 30 so the extruder moves up 30mm after a print.  You can edit that value to change how far the extruder moves.  The value is in mm so if you wanted it to move up 50mm you would simply use the line -
_{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+50, max_print_height)}{endif} F720 ; Move print head up_

Once you have made the change you will then need to save the changes (as you don't want to have to do this for every print).  Just use the little diskette icon next to the name and save it with a custom name.  You have to use a custom name as PS wont let you edit any of the built in system profiles.

Now when you slice a model just make sure that your modified printer profile is selected and it will use the end gcode from that profile, this will move the extruder up however far you have it set in that profile.  You can modify and save as many profiles as you like.

mojeto, 2020-09-10:
I changed End G-code in Prusa Slicer and it works. 
thank you @neophyl detailed steps.
Feel free to close this ticket.

JohnnyDeer, 2020-09-10:
@neophyl thank you for help. Change in PrusaSlicer is better than changes in FW in this case :)

closed 2020-09-10 by JohnnyDeer