prusa3d/Prusa-Firmware-Buddy #874: [ENHANCEMENT] [For newbies] Warn if loaded filament do not match .gcode Prusa Slicer selected one

nekopep, 2020-09-07:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI]

**Is your suggested improvement related to an existing problem? Please describe.**
As a newbie I sometimes splice my stl file with the wrong filament type.
    Example: I slice as PLA while my printer is loaded with PETG.
The printer should emit a warning in this case.
  
**Describe the expected functionality**
  As the PrusaSlicer output filename contains the expected filament name (xxxx_nozzlesize_filament_MINI_timetocomplete.gcode), the mini firmware could detect the filament name into the filename (or even better int gcode with ; filament_type = PLA) and emit a warning on such basic newbies error.
I know that a lot of user will probably use other slicer but in this case, the file will not contains any PLA or PETG term and no warning should be emitted.

nekopep, 2020-09-07:
For completeness, the prusa mini firmware could scan if the gcode come form Prusa Slicer cheking "; generated by PrusaSlicer" and then check that  "; filament_type = PLA" match with printer currently selected filament.
Yes it is totally oriented to the Prusa couple Slicer/Printer, but let's give an advantage to your fanatic newbie user using your software with your printer and easily avoid them some missed print :)

JohnnyDeer, 2020-09-08:
Hi @nekopep this issue is duplicate of #403 and we don't have plan to implement it at this time. I´m closing this one as duplicate. Thank you for understanding.

closed 2020-09-08 by JohnnyDeer

nekopep, 2020-09-08:
Hi @JohnnyDeer ,
I just read #403.

Please reconsider your position, we are speaking about checking data from Prusa inhouse developed slicer working with prusa printer. It is a bit bad not to do a such easy patch that has no impact on compatibilty, you just make your printer firmware more clever with your in house version of your slicer... and save a lot of printing of your user that loves Prusa Brand...
You still keep compatibilty with other slicers, just being more clever with file coming form your Slicer...

JohnnyDeer, 2020-09-08:
@nekopep I understand your intentions and I will mention this suggestion on DEV meeting again.

urholaukkarinen, 2020-10-21:
If matching the filament types is not feasible, could this be solved by a temperature warning? Different filament types have different recommended temperature ranges. When printing is started, we could check if the gcode uses temperatures outside recommended temperature range of the loaded filament.

cnEUL, 2020-10-29:
I have lots of PLA and PETG on the stick and the scrolling windows in the display is very slow. From time ti time it comes up, that i selected the wrong gcode from the stick and then the nozzle clogs. I think this happens everyone from time to time and what sense does it makes, when I can select the filament in the firmware but then in gcode the wrong filament was used (wrong temp). I would like to have a warning at minimum. And by the way ... faster side scrolling on long filenames please (or by setup selectable scroll speed).

fortinmike, 2021-04-06:
@JohnnyDeer Thanks for re-considering this enhancement request. Did you get to add this to the backlog or at least discuss it with the other core devs? I was about to add a new issue requesting this when I stumbled upon this one and some of its duplicates.

In the issue title it says "for newbies", but I think this isn't _only_ applicable to newbies. I have a reasonable amount of printing experience at this point (a couple hundred hours of print time, ~3 months printing every day or two and getting to know PrusaSlicer advanced/expert settings pretty well) and I just made this mistake, wasting filament, time and making my print unfit for purpose. Could I be qualified as a newbie? Maybe. But it's far from impossible for experienced people to make the same mistake one way or another, especially when reusing already-sliced gcode as @cnEUL pointed out.

It's easy to forget to switch filament type in the slicer, especially if you often print with the same filament type (e.g. PLA) and only occasionally with other types of filament. Forgetting to switch filament types can result in weak prints at best and catastrophic failure at worst. It can result in massive waste if the resulting print is not fit for purpose or fails, as well as maintenance headaches if a blob or clog occurs. It's also an easy mistake for people to make in makerspaces, I suppose, which could lead to printers being completely disabled for a while.

As Prusa encourages sharing gcode directly on prusaprinters.org, it would make sense to me to add this additional safeguard. It could also prevent people from mistakenly running other people's gcode with the bad type of filament loaded by mistake.

Could you please consider reopening this issue and possibly removing the `[For newbies]` tag from the title?