prusa3d/Prusa-Firmware-Buddy #894: [BUG] Filament sensor not working

andrelind, 2020-09-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini

**Printer firmware version**-  4.2.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
 The filament sensor doesn't stop the print when running out of filament, I have tried this 3 times now and it fails to stop every time.
The sensor is enabled in the settings but still fails to function

**Video**
https://youtu.be/MJOFrH67QU4

cnEUL, 2020-09-28:
HI, i had some similra in my mini. At mine it was the center screw, it was to tight. Inside is a metall ball wich is liftet in a magnet field. Filament presses the bal down and there is lever which at the end goes in a optical sensor. If the screw is to tight the lever cannot move right. 

Another possibility perhaps is that the two neodym magnets are not justificated right. Plugg not in the right direction or electronic defect.

JohnnyDeer, 2020-10-05:
Hi, @cnEUL thank you for the comment and I am inclined to this description. @andrelind did you tried it?

andrelind, 2020-10-05:
> Hi, @cnEUL thank you for the comment and I am inclined to this description. @andrelind did you tried it?

Yes, I loosened the middle screw and suddenly the printer started beeping when inserting/removing filament :-D

andrelind, 2020-10-05:
Started working properly, sorry for the confusion..!

JohnnyDeer, 2020-10-05:
Then I close this issue as solved. Thank you.

closed 2020-10-05 by JohnnyDeer