prusa3d/Prusa-Firmware-Buddy #895: [FEATURE REQUEST]Add firmwaresupport for temperature compensated bed leveling

stickybyte, 2020-09-21:
I would like to share my results of an analysis of my Prusa Mini.

Problem description
The bed leveling was fine and I startet a print (PLA 215/60°), which was sucessfull. Then I started a second print, and the first layer was more than 1mm too high. Fixed that with live adjustment. So far so good.
One day later I changed to ASA (260/100°) and the nozzle scratched my steel sheet and destroyed it.

I made so many first level calibrations to fix it, bot nothing helps. For me the probe seems to be fine checked with this manual:
https://help.prusa3d.com/de/article/minda-testing_134024 --> All fine!

Then I went to a friend with a working Prusa Mini to do some tests... Our test setup was an RPI with octoprint and bed level plugin.

Results

```
First run with my Mini (cold state)
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.093 +0.179 +0.075 -0.075
Recv: 1 +0.041 +0.186 +0.114 +0.028
Recv: 2 -0.062 +0.078 +0.127 +0.118
Recv: 3 -0.104 +0.059 +0.132 +0.108

Second run with my Mini (preheated from first run)
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.141 +0.187 +0.077 -0.059
Recv: 1 +0.059 +0.197 +0.141 +0.090
Recv: 2 +0.057 +0.193 +0.210 +0.203
Recv: 3 +0.040 +0.179 +0.252 +0.267

Third run with bed temp 100°
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.102 +0.190 +0.085 -0.072
Recv: 1 +0.076 +0.259 +0.212 +0.151
Recv: 2 +0.108 +0.283 +0.307 +0.273
Recv: 3 +0.076 +0.243 +0.315 +0.323

First run with another Mini (cold state)
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.725 +0.488 +0.221 -0.032
Recv: 1 +0.811 +0.613 +0.366 +0.058
Recv: 2 +0.813 +0.582 +0.373 +0.112
Recv: 3 +0.799 +0.544 +0.375 +0.107

Second run with the other Mini (preheated from first run)
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.746 +0.467 +0.201 -0.015
Recv: 1 +0.776 +0.554 +0.305 +0.031
Recv: 2 +0.781 +0.524 +0.319 +0.093
Recv: 3 +0.803 +0.531 +0.352 +0.126

Third run with bed temp 100°
Recv: Bilinear Leveling Grid:
Recv: 0 1 2 3
Recv: 0 +0.679 +0.454 +0.208 -0.042
Recv: 1 +0.790 +0.598 +0.368 +0.066
Recv: 2 +0.781 +0.564 +0.368 +0.116
Recv: 3 +0.761 +0.518 +0.351 +0.101
```
 
Reason and solution
As you can see, my MINDA probe seems to be more and more confused, then warmer it gets. So it is faulty. My replacement is one it's way for free by Prusa. But a better probe must be a focus of Prusa.

This seems to be a problem in general with this sensor and it is definitly not a work around to heat it up before as recommended by the support.

Kranex, 2020-09-28:
@stickybyte I have found that homing my Z axis so that the probe is close to the bed then preheating for 15 or so minutes gets the consistency down enough that my first layer at least doesn't fail. I've accepted that my first layer is going to be awful, but the print doesn't fail and I don't end up scratching the bed. I also monitor the first layer and make a quick adjustment while it's printing the skirt which helps.

It's very annoying though because the minda probe is about the only bad thing about the printer. I'm thinking of trying to see if I can fit a manual bed leveling system or a physical sensor like the one that some enders have because I have seen far better and consistent first layers on cheap enders than I have ever seen on my mini.

leisim, 2020-09-28:
@Kranex My printer is exactly the same. Generally it is an awesome machine but it never gets the first layer right on its own. I also preheat the probe and accept the mediocre result or I just stop the print once I manually adjusted the distance and restart it.
I hope Prusa offers an upgraded probe because I don't think this can be solved with a firmware update. I'd be happy to buy a new sensor if they offer an upgrade.

stickybyte, 2020-09-28:
@all My problems are gone with the exchanged probe ... Temperature doesn't have a hughe impact any more ... just a little live adjustment necessary when switching to ASA (much higher temps)

JohnnyDeer, 2020-10-05:
Hi, this issue is related to #636 #309. We are constantly working on solution, unfortunately it´s long term issue because is touching hardware. FW changes and features wasn't solve it completely.

d-zalewski, 2020-10-05:
P.I.N.D.A V2 sensor cost like £17 and I saw someone retrofitted into the Mini. Unfortunately it requires custom firmware and it breaks the warranty. Hopefully there will be M.I.N.D.A V2 with thermistor available for all Mini owners for free at least as cheap upgrade.

vazmuten, 2020-10-06:
By "constantly working on solution" means you know the problem, it's a hardware one, it could not be fixed via the software and you are working on a hardware solution since when? What are the milestones and planned due date?

leisim, 2020-10-13:
> it could not be fixed via firmware/software and you are working on a hardware solution since when?

What I don't understand is that they already have developed the solution. PINDA. The Buddy board supports a temperature sensor and they just need to adjust the mount and firmware to support the PINDA.

DRKMSTR, 2020-10-21:
> What I don't understand is that they already have developed the solution. PINDA. The Buddy board supports a temperature sensor and they just need to adjust the mount and firmware to support the PINDA.

They appear to be the same size, I have a PINDA v2 (I purchased one) and as soon as they update the firmware I will install it. It's just a temperature table, I'm surprised it's taking an 'act of congress' to get this issue resolved. If they need any help or beta-testing, I'd be more than glad to assist.

If you guys need a case or two of redbull to help solve the problem, let me know, I'll gladly send it over. 😉

vintagepc, 2020-10-25:
I suspect part of it is they want to make sure it's right the first time; unlike software a hardware change is a pretty big commitment - you have to consider the impact not only to new production, existing stock, and also what to do about all the units already sold. In cases of a big issue (think MMU2S kit) they ate the cost and shipped one to every person that had already purchased an MMU2. So while I don't doubt they'd do that if it comes to it for the Mini... you can bet they want to be absolutely certain that it's the best option available before they commit. 

Throw in the issues with the pandemic and getting face time with other people to do R&D... and it's easy to see (even if frustrating) why it might take a while.

d-zalewski, 2020-10-25:
@vintagepc that is fair point however I'm sure they are doing well as everyone at home and wants 3D printer or two. 

Looks like all dev effort recently has been still on MKS3 things like linear advance improvement and Mini gets some really minor (not that important updates in my opinion) like NTP or additional languages support. 

I understand extra languages are better for business as non-English speakers are potential customers but they won't around for long and will join upset group of current ones. 

I would argue dev time would have been better spent on fixing issues like. I'm sure not everyone would agree and but this my view.

DRKMSTR, 2020-10-25:
From experience, @d-zalewski appears to be on to something. 

The Mini has one issue that has yet to be solved - Y crashing at the start of every print (Y axis moves forward, then rearward to zero, however the Y axis is moved all the way forward at the end of each print, causing Y axis motor to skip at the start of every print). 

So it appears they're not actively fixing simple Prusa Mini issues. 

My offer still stands on the redbull for the dev team if they can tackle this :)

vintagepc, 2020-10-25:
The 8-bit dev team is a separate group of folks from the 32 bit team - so it's not an either-or proposition.

I certainly agree things could be more communicative on the 32bit front (disclaimer: I'm fairly new to mini ownership so have not been following any development as closely as say, the MK3S - but my casual impression is it's been fairly quiet in terms of news since the last Mini FW release) but it's not as cut and dried as "just move those people over". The MK3S LA updates (and most other 8-bit updates recently) were mostly the work of community contributors, not people that would otherwise have been working on Mini FW.

DRKMSTR, 2020-10-25:
@vintagepc 

It's not entirely apples and oranges. Last I checked they were moving their 8-bit team over to 32-bit as they transition features and prep for the Mk4 / Mk3SS 🤣 .

I think the real reason we haven't heard much from Prusa's dev team is because they're working on Prusa MEGA / MAX + Prusa Mk4 + Mini Mk2. Odds are they'll do an announcement soon (within the next few months) for those release dates.  

The main issue we all seem to have is the lack of communication. It's frustrating because there are common-sense features: #366 that don't take an extensive amount of time (Give us total print hours plz). 

Add on the lack of being able to implement changes ourselves (or we void the warranty....which many mini users rely heavily upon) and it makes for a rather frustrated userbase.

vintagepc, 2020-10-25:
Yes - mostly just pointing out that getting irritated they worked on MK3S features instead of Mini bugs is a bit misguided given those features were not done internally. 

I too would love to see some sort of feature plan and status dashboard, but alas I don't think we will anytime soon. It's a double-edged sword; given how vitriolic I have seen some discussions get when people feel their pet issue is not a priority over something that is currently being worked on, I can also completely understand why they keep mum on things to the maximum extent possible.

JohnnyDeer, 2020-11-25:
Solution is here, thank you for patience. As you may noticed we released MINI+ with a new Mesh Bed Leveling sensor, the SuperPINDA. SuperPINDA will make calibration of the first layer much more consistent. So all of you with inconsistent first layer issue please contact our Support and we will replace your MINDA for SuperPINDA. I will close this issue soon along with all related. In any new cases of this bug please create new issue to make this repositary clear. Thank you all.

JohnnyDeer, 2020-11-25:
Solution is here, thank you for patience. As you may have noticed we released MINI+ with a new Mesh Bed Leveling sensor, the SuperPINDA. SuperPINDA will make calibration of the first layer much more consistent. You can find it in the e-shop as an upgrade for your MINI to MINI+: https://shop.prusa3d.com/en/original-prusa-mini/1399-original-prusa-mini-to-mini-upgrade-kit.html
If anyone faces issues with inconsistent first layer and you have explored all options first and suspect MINDA to be the cause, please contact our Support with sufficient proof, photos or a video that your issues could be MINDA . I will close this issue soon along with all related. In any new cases of this bug please create new issue to make this repositary clear. Thank you all.

JohnnyDeer, 2021-02-17:
I close this issue, all new cases related SuperPinda please add to #1131 Thank you.

closed 2021-02-17 by JohnnyDeer