prusa3d/Prusa-Firmware-Buddy #932: [FEATURE REQUEST]M600 should move printhead to next location after filament change

LuxuSam, 2020-10-02:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]
**Optional upgrades** - [Filament Runout Sensor]

**Is your feature request related to an existing problem? Please describe.**
  No

**Describe the expected functionality**
  It would be very nice when doing a filament change with M600 that the printhead would move to the next printhead location and not to the old location as this often leaves a small blob of the new filament on that spot. 
When you do a multicolor print with manual filament changes on one layer, the printer always moves the printhead to the old location and then leaves a bit of filament there, and then proceeds to the next location to start printing with the new filament.

jkavalik, 2020-10-02:
I noticed the same. But I believe it is a slicer thing, not a printer firmware issue/feature. PrusaSlicer inserts the M600 directly after the layer change GCode command which means the last location gets stored and then reset after the filament change. Before I stared using the version with M600 GUI support, I inserted it by hand and found out that putting M600 after first XY move of the layer instead directly after the Z move fixes this.

LuxuSam, 2020-10-02:
I do not mean M600 for layer changes, like when you hit the plus sign in the layer view. It is possible to do filament changes like the MMU does, only that you have to do the changes on your own.
I was not sure if it’s a slicer thing as the firmware interprets how the M600 is handled, not the slicer.

jkavalik, 2020-10-02:
Still, the printer does the change exactly where you tell it to - imho it cannot move to the "next" location on its own because that move might be extruding in the original code (you could put M600 in the middle of a originally continuous line). So you can put the M600 between the "move" and the first "extrude" of the differently-colored section.

There possibly might be something like buffering the color change until an extruding move is detected, but with all the retractions and deretractions it seems quite hard to me (not a firmware dev).

andrelind, 2020-10-05:
Would be nice if the printer built up pressure somehow after the change also, the first few centimeters are very thin (if anything at all)

JohnnyDeer, 2020-10-21:
Hi this request is related to PrusaSlicer please create ticket on PrusaSlicer reporistary, thank you.

closed 2020-10-21 by JohnnyDeer