prusa3d/Prusa-Firmware-Buddy #944: [BUG] BSOD on reconnect, if printer is still connected

onkelpit, 2020-10-06:
**Printer type** - [MINI]

**Printer firmware version**-  4.2.1-final+2072

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** USB/Octoprint

**Describe the bug**
The printer was connected via serial with the Octoprint Server. I installed a plugin an restarted the server. The printer still shows that little Octoprint logo in the screen. I clicked on "Connect" in the freshly started Octoprint. A BSOD was shown and the fans started to go on.

**How to reproduce**
1. Connect printer via USB serial
2. Restart Octoprint (via Packageupdater)
3. check if the Octoprint logo stays on the screen of the Mini
4. click "Connect" in freshly restarted Octoprint instance

**Expected behavior**
Eather drop the old session and reconnect or deny the connection.

**G-code**
-

**Crash dump file**
Insidie the zip-archive.

**Video**
-

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5336327/dump.zip)
[image]

JohnnyDeer, 2020-10-07:
Hi @onkelpit thank you for reporting.

Ringous1, 2020-10-08:
What was the plugin? I tried to call the BSOD with these steps unsuccessfully. Normal behavior

d-zalewski, 2020-10-16:
I'm not sure if my issue is related but occasionally when I press disconnect Octoprint from the Mini screen and then go to Calibration menu, it hangs for few seconds, then a beep and watchdog error appears followed by a BSOD and a reboot.

JohnnyDeer, 2020-12-15:
@onkelpit @d-zalewski  unfortunately we are not able to reproduce it. Could you please try it on 4.3.0-RC1?

JohnnyDeer, 2021-02-17:
@onkelpit is this issue still relevant?

onkelpit, 2021-02-17:
@JohnnyDeer i was not able to reproduce it nor had the issue again. Can be closed. If it will happen again, I'ld open a new ticket.

closed 2021-02-17 by onkelpit

JohnnyDeer, 2021-02-17:
@onkelpit thank you for update :)

blejdfist, 2021-03-24:
I have had something similar happen a couple of times, but unfortunately I didn't get a picture of the screen but I did save a crash dump after reset.

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6196540/dump.zip)