prusa3d/Prusa-Firmware-Buddy #976: [BUG] BSOD Firmware 4.2.1

jerometerrier, 2020-10-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [4.2.1-final+2072]

**Original or Custom firmware** [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
  If i want to launch a second first layer calibration, i have a BSOD.

**How to reproduce**
You have to do a first layer calibration, and when the printer ask if you want to restart the caliibration, say yes.
I have a BSOD each time.
So if i want to do two first layer calibration, i have to say no, and to restart a new calibration from the menu.

**Expected behavior**
no bsod, but a new first layer calibration

**Crash dump file**
  
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5395180/dump.zip)

**Video**

[20201017_101152_1.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5395190/20201017_101152_1.zip)

JohnnyDeer, 2020-10-27:
Hi @jerometerrier which language you have the printer in?

jerometerrier, 2020-10-28:
Hi
I'm using the French language.

jerometerrier, 2020-10-30:
Hi @JohnnyDeer , 
i just did the try with English Language, no BSOD.
It's working.

cassou, 2020-11-01:
Hi,
I installed my printer yesterday, and run in the exact same bug.
Crash at the 2nd calibration attempt, each time.

I am also in french.

JohnnyDeer, 2020-11-03:
@jerometerrier thank you for trying i wanted to ask you for it :) 
@cassou ould you please also tri it in english language?

cassou, 2020-11-03:
@JohnnyDeer I just tried in english, it works flowlessy :+1: 

I did one more test in french (just to be sure), crash happened again.
I forgot to mention the firmware : 4.2.1-finale+2072

Let me know if I can do more test ! (I am an embedded software engineer, I was going to try to find the bug myself but I just discovered that this would void my warranty :cry: )

JohnnyDeer, 2021-02-10:
This bug is fixed in 4.3.0. @jerometerrier @cassou thank you for help. I´m closing this issue as solved.

closed 2021-02-10 by JohnnyDeer