prusa3d/Prusa-Firmware-Buddy #977: [BUG] Hotend cooling fan insufficient supply voltage

jack-aaron-cooper, 2020-10-18:
**Printer type** - MINI
**Printer firmware version** -  4.2.0 - final+2067

**SD card or USB/Octoprint**
USB

**Describe the bug**
The hotend cooling fan on boot will have the correct 5V (4.99V), however, when preheating or printing it will have approximately 2.2V. As a result of this the fan will not spin without a push start and at that voltage the fan will not spin fast enough to provide enough cooling to the hotend.
  
**To Reproduce**
- Check hotend cooling fan voltage on boot.
- Check hotend cooling fan voltage on preheat or printing.
  
**Expected behavior**
When preheating or printing the voltage should be 5V, when it is above the fan activation temperature

**Images**

Boot
[image]

Preheating/Printing
[image]

Measurement points
[image]

dominickuznik, 2020-10-18:
I have the same problem with a clone, at first I thought that the firmware recognizes a "foreign" fan when booting and does something stupid :)
when my original came, I saw how much air it promotes.
you can take the 5 v from the filament sensor slot, I did that to cool the buddyboard :)

x10102, 2020-10-18:
My cooling fan works just fine during printing so I assume the voltage is okay as well. It's probably a problem on your board.

dominickuznik, 2020-10-18:
Although the correct 5v is present when booting - if there is a problem with the board it will be replaced by prusa.

jack-aaron-cooper, 2020-10-18:
I'm pretty sure it is a preset PWM value in the firmware, to keep the fan quiet.

jack-aaron-cooper, 2020-10-19:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/95447c177e0d3301713266200ed2620fe10551ae/src/common/config_a3ides2209_02.h#L83

JohnnyDeer, 2020-10-21:
Hi @jack-aaron-cooper as you wrote, the PWM value in the firmware is to keep fan quiet. So this wont be changed. I close this issue, thank you for understanding.

closed 2020-10-21 by JohnnyDeer