prusa3d/Prusa-Firmware-Buddy #981: [BUG] Pausing a print while mesh bed leveling is running might damage your printer

polygonal, 2020-10-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
When you select pause after starting a print and while mesh bed leveling is running (before the first line has been printed), the nozzle went wild and after some erratic movement it crashed into the bed. When I noticed it I pressed reset to prevent more damage. Printer is still working, but the bed is now slightly misaligned since It seems some bending occured. The machine is new and it worked perfectly fine before, no issues with self test/pinda etc.

**How to reproduce**
Start a print. Wait until mesh bed leveling is running, then pause printing.

**Expected behavior**
Abort mesh bed leveling and repeat when user resumes print

**G-code**
I have no gcode or crash dump file

jakubmirejovsky, 2020-11-09:
Same here, i attach a video:

Prusa MINI, firmware 4.2.1

Right after Mesh Bed Leveling (under Calibration), the nozzle goes down and touches and bends the hotend.
Video here: https://www.youtube.com/watch?v=4gR8gOi_ESg

To Reproduce
It happended to me when I:

started printing (went for 15 minutes)
paused printing for 6 hours
resumed printing
stopped printing after 5 minutes
Z axis moved up
first layer calibration
Mesh Bed Leveling - here the problem was spotted. I tried it again with my camera on (see youtube link).
After resetting the printer, it went just fine.

Expected behavior
I expect nozzle not to touch the surface at all.

G-code
I am attaching the gcode, but I don't think it will be useful for debugging:
https://www.dropbox.com/s/63aloo0s4qaw169/Spi_FRA_public_up_cov_0.25mm_PLA_MINI_3h27m.gcode?dl=0

Video
https://www.youtube.com/watch?v=4gR8gOi_ESg (0:43)

Picture
https://www.dropbox.com/s/4r8s9gdie9yap9v/2020-10-31%2020.29.42.jpg?dl=0 (PEI damaged)

JohnnyDeer, 2021-08-10:
@polygonal, I tested this issue on 4.3.2 and bug don't occurs. Could you please try it too?

polygonal, 2021-08-11:
Can't reproduce it either 👍

JohnnyDeer, 2021-09-02:
@polygonal thank you, then i'm closing this issue.

closed 2021-09-02 by JohnnyDeer