prusa3d/Prusa-Firmware-Buddy #990: BSOD after a Watchdog reset on 4.2.1

JohnnyDeer, 2020-10-22:
I'll bite. Got a BSOD with the same message after a Watchdog reset on 4.2.1; this was the second or third layer of the print, if that matters. Will retrieve dump from USB after print finishes... repeating the print did not make it recur, so it does not seem to be G-code specific.

[image]

_Originally posted by @vintagepc in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/763#issuecomment-711063459_

vintagepc, 2020-10-22:
> Thank you again for your prompt answer. After some investigation I need to ask you another questions just to confirm what could happened:
> 
> Did the BSOD occur immediately after the printer startup? (< 20 seconds)

No, this was after it had been on for a bit; this was a second or third print after poweron. 

> Last what you did was press the knob

Correct - I inserted the USB stick with a newly sliced file, and pushed "Print" on the confirmation screen that pops up.

> Did you do anything else? (plug/unplug the USB stick, ethernet cable,...)

No - I remember I watched the first layer stick down, a little bit of the second layer, and then walked away; A few moments later I heard the printer stop moving, looked over,  and the watchdog error was on screen.

marekmosna-prusa3d, 2020-10-23:
Hi @vintagepc 

Such a bug will be fixed with the next version of Firmware-Buddy. I hope it will not be disturbing you and your printing too much. If anything happened please let us know. We appreciate your attitude and effort that help us make the firmware better.

Thank you
Development team

vintagepc, 2020-10-23:
Thanks - so far no recurrences yet, but good to hear you've got something in the works for the next version!

inferior-minion, 2020-12-11:
I've gotten at least 4 resets on my Mini running 4.2.1. I updated as soon as it came out. I've attached the dump file from the latest one. Please help, this is very annoying and has been ongoing for a while.
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5680965/dump.zip)

JohnnyDeer, 2020-12-14:
This bug is fixed in 4.3.0-RC1. I'm closing this issue.

closed 2020-12-14 by JohnnyDeer