prusa3d/Prusa-Firmware-Buddy #997: [FEATURE REQUEST]Hotend fan PWM setpoint modification

PeterDHabermehl, 2020-10-23:
I would like to run the hotend fan at higher speed. I installed Bondtech heatbreak and a copper heat block for better high temperature printing performance, now I would like to raise the hotend fan speed.

Building my own firmware would be an option, I'm willing to break the appendix, but where to change the firmware to run the fan at 75% or 100%?

mattderen, 2020-10-25:
This is a function of the slicer not just the firmware. In PrusaSlicer, you have to disable auto fan speed and lock it in at 100%. In Simplify3D you can simply adjust your printer profile.

If you like to edit gcode, just find and replace all the M106 commands with `M106 S255`.

PeterDHabermehl, 2020-10-25:
Thanks for the answer but I'm talking about the hotend fan. NOT the print/part cooling. I'm well aware of M106 and how the part cooling settings work.
The control of the hotend fan is hardcoded in the firmware and actually somewhat different from normal Marlin code.

mattderen, 2020-10-25:
That should be easy to find, actually. Give me a sec.

mattderen, 2020-10-25:
As a second note, the hotend fan is usually wired directly to the 12/24V rail, but it wouldn't surprise me if Prusa leverages PWM for it as well.

PeterDHabermehl, 2020-10-25:
The Mini hotend fan is PWM controlled. And I know that a fw hack is used to enable HW PWM control instead of the usual Marlin sw PWM.
The Delta fan used is a high peformance device and operated at (probably) 50% PWM only for noise reduction. Since I want to upgrade an existing Mini I would like to plug the replacement fan directly into the HE fan connector of the Buddy Board. There's no simple way of connecting directly to the PSU.

mattderen, 2020-10-25:
Yup on the controlled fan: https://github.com/prusa3d/Buddy-board-MINI-PCB/blob/master/rev.1.0.0/BUDDY_v1.0.0.pdf (page 2)

There is a value [here](src/common/marlin_vars.h) that is of interest. (MARLIN_VAR_FAN1_RPM)

There are also Min/Max PWM pulse settings that I just found, and lost. Give me another 5.

mattderen, 2020-10-25:
@PeterDHabermehl https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/089b442e16389becfe7d3583788e17ae7eb15e6f/src/common/config_a3ides2209_02.h maybe?

mattderen, 2020-10-25:
Also, you may need to look into if watchdog has any checks regarding hotend temperature timing checks, btw. That is some serious speculation on my end, however. Still, its just a thought.

mattderen, 2020-10-25:
Cool.  Thanks for taking me down this little path of source code spelunking and hope I was able to help. Cheers!

PeterDHabermehl, 2020-10-25:
Thanks for taking the dive into the source code. I'll see what I can figure out.

BerenV, 2020-11-04:
@PeterDHabermehl I ended up building a custom hotend fan controller using an ATTiny microcontroller to control the speed of the Noctua fan I installed. It's a lot quieter than the stock Delta fan and I have a little knob to adjust the speed. The airflow isn't as good as the Delta fan but it sure is a lot quieter and I haven't run into any issues just printing PLA. I might have to turn it up to prevent clogs when printing PETG or PC filaments though.

I don't know why they didn't implement temperature based speed control for the hotend fan, but it would be a nice feature to be able to easily modify. I was too lazy to dig into the source code and compile my own version so I went the hardware way. Good luck with your modification!

[image]

Lino77, 2020-11-07:
@PeterDHabermehl https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/304

blerrgh, 2020-11-15:
I think a lot of people (myself included) would like to change the hotend to a silent noctua with a slower rpm that needs the full 5Vdc.  At the very least exposing the PWM constant as an advanced setting would help in those scenarios.

PeterDHabermehl, 2020-11-15:
@BerenV 
Thanks for that one - where did you connect to the +5V to supply your controller?

@JohnnyDeer I would second @blerrgh 's request. But for the moment It would really great to know where to modify the fw to get 5v to the fan.

BerenV, 2020-11-15:
@PeterDHabermehl actually the fan is supplied with +5v all the time and the fan's ground is PWM'd to get a certain speed. I had to run a ground wire from the board to my ATtiny setup to power it. Here's a schematic of the way I did it, and if anyone wants the code just let me know (it's super simple)

[image]

A-Bielefeld, 2021-02-15:
I second what's been said before, I'm also switching to a Noctua fan, and it seems to be a popular upgrade since the original fan is so loud. I'd definitely like to see an option for controling the PWM in software, options, or GCode, there's no reason this feature should be missing.

zoltan-l, 2021-09-13:
Just an explanation, the below statement means the maximum speed.

static const uint8_t FANCTL1_PWM_MAX = 50;
As the PWM pulses are 20ms long the 50*20=1000ms which is the full dutty cyclus period lasting 1 second.
So any trial to increase this number does not make any effect. When you run the  FANTest, you can see the hotend fan max speed is arround 8000 RPM, which is really the full speed of the current hotend FAN on 5Volts.

Regarding the Noctua, I guess its maximum speed is lower than 8000 rpm, and I am not sure, they have a turbine.

OktapBangLab, 2021-09-26:
Apparently, FAN1(Hotend/Heatbreak Fan) PWM set point is hard coded in hwio_buddy_22009_2.cpp at line 576 fanctl_set_pwm(1, ulVal ? (100 * _**50**_ / 255) : 0);
where bold italic '50' is the PWM set point setting. When increased to 80, RPM increased from approx. 3800 to approx. 5500.

MinnesotaMean, 2022-03-29:
It seems fan PWM control has changed as changing the variables in hwio_buddy no longer works.

Harcos98, 2022-07-03:
Hello to anyone! Is there any news about this topic? Here in Italy the climate is quite hot and the hotend fan is struggling to keep the heartbreak decently cool. I'd really love to be able to control fan's speed via display instead of having to recompile the firmware. La there a way to bring prusa's attention on this topic?

SnickCH, 2022-07-11:
+1 - I also need to speed up the fan speed (I don't care if the stock hotend is louder or will break after 1 year). I need to better cool the heatbreak. It would also make it easier to replace it with a better fan

Harcos98, 2022-07-11:
Also, it should be really simple to implement for prusa's developers. Would literally take half day of work

SnickCH, 2022-07-11:
Probably it will be very easy to fix.

If I have to guess: Prusa promised so many things with the mini (like Prusa connect) and there are so many issues, that even a half day of work will probably be to much at the moment (half a day per issue-> for all the issues = 1 year of work for a developer only to fix it).

I hope that Prusa proofs me wrong :) and it will soon be implemented.

closed 2024-07-04 by github-actions[bot]