prusa3d/Prusa-Firmware-Buddy #1000: Issue with time change BST - GMT ?

Gordontgopher, 2020-10-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** USB

**Describe the bug**
Early this morning UK changed from BST to GMT. I just turned on my printer and started a 10 min print, the printer showed an end time of 1hr 10min from now but connect local showed the correct end time of just 10 mins later. It appears that the printer did not recognise the change in 'offset' from GMT / zulu.

**How to reproduce**
See above!

**Expected behavior**
Printer to recognise such time changes.

**G-code**
Not g-code related

**Crash dump file**
no crash = no dump file

**Video**
  Please attach a video. It usually helps to solve the problem.

JohnnyDeer, 2020-12-16:
Hi @Gordontgopher, Mini don't know about time zones and time changes, have just raw clock - will show time depended on how you set it.

Gordontgopher, 2020-12-16:
So we have to change the clock twice a year to reflect DST changes?

closed 2020-12-16 by Gordontgopher