prusa3d/Prusa-Firmware-Buddy #1015: [BUG] Manual Axis Movement Problem - Extruder caught in infinite movement loop

DRKMSTR, 2020-11-01:
**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB Drive

**Describe the bug**

I preheated the bed and moved the "E" axis (extruder) by 15mm, to clear out all the old tempered filament from a bunch of preheat cycles. After about 5 minutes, it was still extruding filament and I had to manually restart the printer to stop it.

**How to reproduce**

Unsure, I was not able to reproduce it. It was preheated to PETG, then the "E" axis was told to manually move 15mm (or whatever to the soft limit (where you can't scroll any more), then a few mm more as it extruded)

**Expected behavior**

I expected it to extrude for a minute, maybe two to clear out the old tempered filament.

**G-code**

N/A

**Crash dump file**

N/A

**Video**

N/A

JohnnyDeer, 2020-11-18:
Hi @DRKMSTR unfortunately we are not able to reproduce it. Please wait till new FW will be released and try to reproduce this issue. Thank you.

JohnnyDeer, 2020-12-15:
@DRKMSTR 4.3.0-RC1 is out could you please try it?

DRKMSTR, 2020-12-15:
> @DRKMSTR 4.3.0-RC1 is out could you please try it?

Will do as soon as I get some time. 

I need to install the new probe and a plated copper nozzle first.

JohnnyDeer, 2021-02-17:
@DRKMSTR is this problem still actual?

JohnnyDeer, 2021-05-13:
I'm closing this issue because of no response.

closed 2021-05-13 by JohnnyDeer