prusa3d/Prusa-Firmware-Buddy #1030: [BUG] Filament sensor fails to pause OctoPrint

BerenV, 2020-11-04:
**Printer type** - MINI

**Printer firmware version**-  4.2.0

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - Octoprint

**Describe the bug**
 I just had a relatively long print get ruined by a failure of the printer to pause when it ran out of filament. I have the sensor installed and enabled, it is reading the state correctly but it didn't pause Octoprint

**How to reproduce**
Try printing a longer print in Octoprint and cutting the filament I guess...
  
**Expected behavior**
  The printer should pause when it runs out of filament!

Anyway, I'm updating my firmware to 4.2.1 but I didn't see anything about this in the release notes. It would be awesome if this could be fixed but for now I guess my filament sensor won't be doing me any good since I print almost exclusively over Octoprint. I've also had a print (luckily a smaller one) fail to pause and initiate the color change two times. However, I've also had it work just fine so I'm not sure what to think. I'm using PrusaSlicer and mostly stock profiles.

[image]

Thanks!

JohnnyDeer, 2020-11-04:
Hi @BerenV this is duplicate of #739. I close this issue.

closed 2020-11-04 by JohnnyDeer

BerenV, 2020-11-04:
Ok thanks!