prusa3d/Prusa-Firmware-Buddy #1032: Auto-home combined with bed only pre-heat

RatPatrol1, 2020-11-04:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]

Unless this feature is already built in and have not found yet, a useful macro would be a button to auto-home and bed pre-heat to the already loaded filament type. It would then preheat the bed only and auto-home. You would have consistent pre-heating for the MINDA sensor.

 

I do this manually, except I go into settings, temperature, bed temp to 80 (for PETG), then back to calibration, auto-home.  I generally let it pre-heat while I'm working on slicing, or at least 5-10 minutes to warm up the MINDA. This eliminates the problems with the bed sensor lacking thermal compensation. 

This could easily be written into startup code, but it is nice to have the printer pre-heated while working on the slicer so the printer is ready to go.

neophyl, 2020-11-05:
Why not just create several preset 'prints', one for each of your materials that you commonly use and put them on the usb stick ?  That way you could just 'print' "Preheat PLA" or "Preheat PETG".  Much simpler than changing firmware and you get the precise temperatures you want.

RatPatrol1, 2020-11-05:
I suppose that could be done. 

Do you happen to know if the Mini can accept logic/calculation commands? I could just write a custom Gcode to put the preheat on a timer.

neophyl, 2020-11-05:
As far I I know none of the home printers do any sort of calculations.  They are pure dumb beats that just follow a set of instructions.  All the calculations are done at slice time.
Of course if you are running something like Octoprint or running Klipper then those systems spool out the g-code during the print and can do extra stuff but none of the current Prusa printer do that (I may be wrong about the mini but I don't think so).

closed 2024-07-04 by github-actions[bot]