prusa3d/Prusa-Firmware-Buddy #1036: [BUG] Blob after colour change/M600

vintagepc, 2020-11-06:
**Printer type** - [MINI]

**Printer firmware version**-  4.2.1

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**

When using PrusaSlicer with a colour change, Mini produces a blob at the place it starts extruding again. 
This appears to be caused by the fact the sliced gcode expects the filament to be retracted after the M600 is complete:

```
M600
G1 X54.508 Y77.639
G1 Z2.100
---> G1 E3.80000 F2400.00000
```

**How to reproduce**
Insert an M600 on a print with small details, e.g. an RPG mini with the base in one colour and the figure as another colour. Observe the first location after the print resumes is massively overextruded, even if the nozzle is clear of all filament after the colour change process. 

**Expected behavior**
M600 should retract the filament when complete. (This fix may need to be coordinated with PrusaSlicer - not sure how you want to handle it)

**G-code**

[Archer_0.1mm_PLA_MINI_3h18m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5501917/Archer_0.1mm_PLA_MINI_3h18m.gcode.zip)

hemdrup, 2020-12-19:
Agree, also get a blob each time.

MTBehnke, 2021-04-14:
I have the same blob issue related to the same line of gcode.  Example photo included below showing the blob around the letter 'R' (white with black letters).  When I comment out the line I don't get the blob (black with white letters)
```
;COLOR_CHANGE,T0
M600
; printing object Blob Test.stl id:0 copy 0
G1 X82.563 Y68.958
G1 Z2.150
;G1 E1.40000 F3000.000  <--- Commented out for black with white letters example.
M204 S600
```
The small blob and string from the upper left corner (starting position after color change) to where it started printing the letters is a minor issue and easily removed.  Also, the gap in the letter 'M' shows up as a gap in PrusaSlicer as a result of the small letters (2.5mm letters), not a printer issue.

[image]

Note:  Prusa MK2S, firmware 3.2.3, PETG, 0.25mm nozzle, 0.15mm layers, PrusaSlicer 2.3.0.

vintagepc, 2021-04-14:
FYI this is the repo for the Prusa Mini firmware. You are looking for https://github.com/prusa3d/Prusa-Firmware

MTBehnke, 2021-04-15:
Thanks for the heads up.

 

This seems much more likely to be a slicer issue as it’s gcode related (manual edit of gcode solved issue).  I created a new issue in the PrusaSlicer repo.

 

Thanks,

Mike

JohnnyDeer, 2021-09-13:
This issue is solved by 4.3.3-RC,  please try it and let know back so this issue could be closed.

JohnnyDeer, 2021-10-18:
I'm closing this issue as solved.

closed 2021-10-18 by JohnnyDeer